espotek-org/Labrador #4: Documentation and links to software

IrregularShed, 2017-07-08:
My Labrador arrived this morning. Hooray! However, I can't find any documentation that explains what the various headers are on the production version, and I can't find a Windows installer. I'm not sure where to go from here 😞

timhughes, 2017-07-10:
I have managed to get the interface running but only briefly. 

I am running Linux and had the interface startup but then crash shortly afterwards. 

These are the steps I used.

1. Git clone this repo
    `git clone https://github.com/EspoTek/Labrador`

2. Change directory to the desktop interface dir.
    `cd Labrador/Desktop_Interface/`

3. Set my Library path and run _./bin/Labrador_
    `LD_LIBRARY_PATH=$LD_LIBRARY_PATH:./build_linux/libdfuprog/lib/:./build_linux/libusb/ ./bin/Labrador `

timhughes, 2017-07-10:
If you are running Windows or Apple then there appear to be binaries under `Labrador/Desktop_Interface` but you may need to set some library paths

Streuphy, 2017-07-10:
Hello,
I'm running Arch Linux and having exactly the same issue.

It looks like the usb device can't be detected by libusb.

It's not too clear now but lsusb does report an Atmel ATxmega32A4U DFU bootloader but there is still error in the code I believe.

I start the binary slightly differently:

1 - I'm not pointing ny LD LIB to the provided libusb but rather use the default system one.

2 - running in sudo for R/W permissions to the /dev/bus/usb object.

I'm going slightly further but still end up with an error...

`sudo LD_LIBRARY_PATH=$LD_LIBRARY_PATH:./build_linux/libdfuprog/lib/ ./bin/Labrador`

I'm not entirely sure that the host (my PC) is at all talking to the board.

If anyone can claim success connecting the board from a linux host.

Do we have to compile libdfuprog for our distro ?

maks, 2017-07-10:
@Streuphy you shouldn't need to have the board in DFU mode unless you are trying to flash new firmware code onto it. Likewise touching the permissions for /dev/bus/usb is **not** the correct way to do it.
What you will need to do is find the vendor and product ids of the boards using lsusb (sorry I cant tell you what they are as I only have a protoytpe not the final board).
Once you have that you will need to add that to your udev rules, so something like:
```
sudoedit /etc/udev/rules.d/10-labrardor.rules
```
and then in that file add:
```
SUBSYSTEM=="usb", ATTR{idVendor}=="abcd", ATTR{idProduct}=="0006", MODE="0666", GROUP="plugdev"
```
then
```
sudo udevadm control --reload-rules
sudo service udev restart
```

EspoTek, 2017-07-11:
Sorry, just saw this.  The repo is a bit esoteric.  More than it should be.  I'll need to tidy it up a lot.

Bin builds are all available here:
https://drive.google.com/folderview?id=0B7U0ulRLHf8cRVBkeFc2SHpUOGs

The Android build will be on Google Play, and I'll talk to the F-Droid folks as well for those who don't like Google on their phones.  Now I just need to work out how to setup GitHub notifications...

closed 2017-07-11 by EspoTek

IrregularShed, 2017-07-11:
@EspoTek thanks Chris, that works wonderfully on my Windows box!
[image]

EspoTek, 2017-07-11:
Good to hear!!

---- On Tue, 11 Jul 2017 19:29:44 +1000 notifications@github.com wrote ----

@EspoTek thanks Chris, that works wonderfully on my Windows box!