espotek-org/Labrador #5: Segmentation fault on Fedora Linux

timhughes, 2017-07-12:
Still having a few difficulties on the Linux build on Fedora. 

I have converted the .deb package to a rpm like this:

```
mkdir ./labrador
dpkg-deb -R labradeb.deb ./labrador
fpm -s dir -t rpm  -v 0.9.1 -n labrador -p labrador-VERSION-ARCH.rpm -d libusb -C ./labrador etc usr
```

/var/log/messages
```
Jul 12 20:10:30 titanium kernel: usb 1-1.1: new full-speed USB device number 19 using ehci-pci
Jul 12 20:10:30 titanium kernel: usb 1-1.1: New USB device found, idVendor=03eb, idProduct=ba94
Jul 12 20:10:30 titanium kernel: usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Jul 12 20:10:30 titanium kernel: usb 1-1.1: Product: Labrador
Jul 12 20:10:30 titanium kernel: usb 1-1.1: Manufacturer: EspoTek
```

Then starting Labrador
```
 thughes@titanium [0] $ /usr/share/espotek/labrador/Labrador
QObject::connect: No such slot bufferControl::signalGenIn(bool)
QObject::connect:  (sender name:   'signalGenGroup_CH2')
QObject::connect:  (receiver name: 'bufferDisplay')
QObject::connect: No such slot timedTickBox::resetTimer(bool)
QObject::connect:  (sender name:   'lockPsuCheckBox')
QObject::connect:  (receiver name: 'lockPsuCheckBox')
Attempting to open "/usr/share/espotek/labrador/waveforms/_list.wfl"
nameBuffer =  Sin

nameBuffer =  Square

nameBuffer =  Triangle

nameBuffer =  Sawtooth

nameBuffer =  DC
newName =  "Sin"
opening /usr/share/espotek/labrador/waveforms/Sin.tlw
128
Length =  128
Divisibility =  4
List loaded!!
Attempting to open "/usr/share/espotek/labrador/waveforms/_list.wfl"
nameBuffer =  Sin

nameBuffer =  Square

nameBuffer =  Triangle

nameBuffer =  Sawtooth

nameBuffer =  DC
newName =  "Sin"
opening /usr/share/espotek/labrador/waveforms/Sin.tlw
Length =  128
Divisibility =  4
List loaded!!
Making USB Driver invisible!!
pipeID 0 = 129
Generic Usb Driver setup complete
unixUsbDriver created!
driver =  unixUsbDriver(0x2328940)
axes =  QCustomPlot(0x21f5f90, name="scopeAxes")
ReinitUsb Stage 1 complete
CHECKING CONNECTION!
Entering unixUsbDriver::usbInit
Libusb context initialised
Device found!!
KERNEL DRIVER INACTIVE
Interface claimed!
unixUsbDriver::usbSendControl FAILED with error LIBUSB_TRANSFER_TIMED_OUT
BOARD IS RUNNING FIRMWARE VERSION 0x0001
unixUsbDriver::usbSendControl FAILED with error LIBUSB_ERROR_PIPE
FIRMWARE VARIANT = 0x01
EXPECTED VARIANT = 0x02

FIRMWARE MISMATCH!!!!  FLASHING....

FLASHING  /firmware/labrafirm_0002_02.hex
unixUsbDriver::usbSendControl FAILED with error LIBUSB_ERROR_PIPE
BA94 closed
argc = 4
argv[0] = dfu-programmer
argv[1] = atxmega32a4u
argv[2] = erase
argv[3] = --force
attempting to call main()
dfu-programmer: no device present.
Segmentation fault (core dumped)
```

timhughes, 2017-07-12:
Looking through the trace i think I have managed to work it out 

I installed dfu-programmer and then flashed the firmware manually

```
dfu-programmer atxmega32a4u erase
dfu-programmer atxmega32a4u flash /usr/share/espotek/labrador/firmware/labrafirm_0002_02.hex
```

Then I needed to power it off and on again before running  /usr/share/espotek/labrador/Labrador

closed 2017-07-12 by timhughes

burtonscomputers, 2017-07-12:
Yup I was just writing to suggest that is what was needed.
Hope it all works better for you. Glad to hear a manual update worked. Chris Esposito has been working had to get that to work as part of the install, but that would not have happened with your ,deb conversion.