espotek-org/Labrador #7: unixUsbDriver::usbSendControl FAILED with error **UNKNOWN**

GrahamCobb, 2017-07-16:
I am trying to get my Labrador working -- I am using Debian Testing (Buster).

I have installed the .deb, and set up udev. The first time I ran the software it upgraded the device firmware (and it now reports the correct firmware version is installed).

The software runs, and says "Device Connected". However, nothing appears in the 'scope screen. Meanwhile, the software is printing, once per second:

unixUsbDriver::usbSendControl FAILED with error **UNKNOWN**
debug123
trfcnt0 = 192
trfcnt1 = 1
medianTrfcnt = 200
outOfRange =  2
counter =  12015
calValNeg =  49
calValPos =  53
CALA =  49
CALB =  14
dma_ch0_cnt =  0
dma_ch1_cnt =  0

(some of the numbers change a bit).

Is this error significant? Or am I just doing something completely stupid?

GrahamCobb, 2017-07-16:
Sometimes the reported error changes from Unknown to LIBUSB_ERROR_PIPE. I haven't worked out what made it change.

EspoTek, 2017-07-17:
Hi Graham.  The error messages are a bit misleading.
"unixUsbDriver::usbSendControl FAILED" happens whether or not the transfer is successful.  On fully working systems you'll get them whenever communicating over the standard control endpoint.  All scope data comes over the isochronous endpoint.
The other output all just relates to internal clock calibration (looks like it's working fine, by the way!)

What's most likely the case is that your system can't reserve the bandwidth needed to connect to the isochronous endpoint.
See instructions below:
"Move Labrador to a different USB port that does not share bandwidth with devices that stream constantly (in proper terminology: isochronous/interrupt devices).  This includes things like keyboards, mice and speakers.

Some suggestions:
If you’re using a hub, connect your other devices to the hub and give Labrador its own dedicated port.
Plug your other devices into your USB2.0 ports, and Labrador into the USB3.0 or eSata ports (or vice versa).
Devices that stream constantly are the most likely to compete with Labrador for bandwidth.  If you are forced to share a port with Labrador, share it with something like a USB flash drive or card reader that does not stream constantly.
If need be, disconnect unneeded USB devices.

Your Labrador board should start working."

Also of note, if you're on an older system (circa 2011), USB3.0 ports might simply not be compatible with the isochronous endpoint.
There is a low-bandwidth mode that doesn't compete with other USB devices for bandwidth, but it's not fully compatible with Mac or Linux (~10% chance that it simply fails to connect at the initial stage, and then all packets from then on are corrupted.  This is not fixable without writing kernel-level driver code.).

GrahamCobb, 2017-07-17:
Thanks. I tried a few different USB ports and found one that worked!

closed 2017-07-17 by GrahamCobb