espotek-org/Labrador #11: Labrador Documentation in Markdown?

gmacario, 2017-07-27:
Hello @EspoTek ,

I received my Labrador today and I have just started having fun with it!

I think it could be useful to have the "Labrador Initial Documentation" converted from Google Docs to Markdown and placed here on GitHub - either inside this repository or - maybe even better - in a separate one, like Labrador-Docs. That would IMHO simplify commenting / contribution (simply via PR), etc.

Anyway thanks a lot for such awesome project!

EspoTek, 2018-02-05:
So going through the open issues at 3AM and this looks like a great idea.
The binary releases have been navigated away from Google Drive, so this is making even more sense.

Do you know the cleanest way to do the conversion?  I looked up a few extensions, but apparently each of them has their own unique little bugs.

gmacario, 2018-02-05:
Hi @EspoTek,

I am glad you supported the idea.

I have never attempted an automated conversion from Google Doc to markdown, but in the past I have used [pandoc](http://pandoc.org/) to do semi-automated, one-time conversions from markdown to HTML and Atlassian Confluence

You may give it a try at https://pandoc.org/try/

My 0.02 EUR

EspoTek, 2018-02-06:
Unfortunately pandoc didn't work.
However, there was a Google Docs extension that spat out the documentation at the end of the post.
Of course, I'd need to do some manual editing.  I'm also wondering whether markdown is the best format, as opposed to, for example, a PDF file.  There's better integration with GitHub, of course, but the user has to read it in-browser, and it isn't available offline unless they download the whole web page.

Here is what the auto-generated file looks like:

Labrador Documentation 

(Work in Progress)

Chris Esposito <[admin@espotek.com](mailto:admin@espotek.com)>

[[TOC]]

# Licence

This document is licensed under the Creative Commons 4.0 BY-SA licence.  You are free to download it, share it, and use it for any purpose (including commercial purposes), as long as any edits you make are shared back to the community, and that proper credit is given (a link back to espotek.com will suffice).

[http://creativecommons.org/licenses/by-sa/4.0](http://creativecommons.org/licenses/by-sa/4.0)

# Introduction

Welcome to the initial documentation for the EspoTek Labrador!

Please note that it’s currently in a WIP state.  If you have any suggestions, please right-click and leave a comment!  It’s harder than you’d think to write a guide for something you invented yourself, so any questions or suggestions help!

Note that the "Using the Hardware" section is mainly based on confused emails I got from software people.  If you’re familiar with electronics, you can safely skip it.

If you’re familiar with lab equipment, you can also skip most of the "Using the Interface" section, but I do recommend reading the section on “buffers” and everything in bold!

Apart from that, enjoy!  Let’s move onto the setup stage.

# Interface Installation

Binary releases for all 3 platforms are available here:

[https://drive.google.com/open?id=0B7U0ulRLHf8cRVBkeFc2SHpUOGs](https://drive.google.com/open?id=0B7U0ulRLHf8cRVBkeFc2SHpUOGs)

The Windows .exe installer contains both 32 and 64 bit versions.

There are separate .deb files for 32-bit and 64-bit Linux.

The OSX software is 64-bit only.

It’s recommended that you install the Linux software using the install script (not by double-clicking the .deb).  If you do not use the install script, you will need to restart after installing the .deb.

The software on all platforms is in a stable state (as far as I’m aware), but there may be some minor bugs present!

# Using The Hardware

## Pinout

[image]

Yeah, it’s not very neat.  Sorry.

If you don’t know what AC coupling is, please don’t use the AC-coupled pins.  Use the DC-coupled pins instead as they don’t alter your signals.

Currently asking some folks on Fiverr to make a nicer diagram for me.  Will upload when it comes through.

## Connecting Your Labrador to a Breadboard

[image]

(image credit: Gianpaolo Macario)

You can connect your Labrador board as shown above.  Power supply pins should go in the side rails to prevent any shorts.

Of course, if you don’t have a breadboard handy, the board can also be used without one.  But where’s the fun in that?

[... 292 lines cut ...]
* If you’re using a hub, connect your other devices to the hub and give Labrador its own dedicated port.

* Plug your other devices into your USB2.0 ports, and Labrador into the USB3.0 or eSata ports (or vice versa).

* Devices that stream constantly are the most likely to compete with Labrador for bandwidth.  If you are forced to share a port with Labrador, share it with something like a USB flash drive or card reader that does not stream constantly.

* If need be, disconnect unneeded USB devices.

Your Labrador board should start working.

Some Mac OSX users have reported driver-related issues that prevent the Software Interface from communicating with the Labrador board.  To fix it, install [homebrew](https://brew.sh/) and then run the following two commands from the terminal:

brew update

brew install libusb

After resetting your computer, it should work.

### All other problems.

If you’re having any other issues with your board, I’m happy to help.

The best thing to do is to post an [issue on the GitHub page](https://github.com/EspoTek/Labrador/issues), that way any solutions will be posted publicly.

The second best thing to do is to send me an email.  I can be found at [admin@espotek.com](mailto:admin@espotek.com).

EspoTek, 2018-02-06:
OK, I give in.  :)
https://github.com/EspoTek/Labrador/wiki

closed 2018-02-06 by EspoTek