espotek-org/Labrador #12: Do record or snapshot do anything yet?

rp10007, 2017-09-15:
I'm using the original mac dmg download. clicking "record" in the file menu turns the checkmark on or off, but I can't see any file being generated. Same for "snapshot". There appears to be a slight hiccup in display when selecting those, so maybe something is happening, but it's not obviously visible anywhere.

thanks.

EspoTek, 2017-09-16:
There was an update released to the software in mid-July.
If you're using that one, it should save all snapshots and recordings to Documents/EspoTek.

A snapshot saves a CSV file of the samples on-screen at the time, whereas a recording will save every single sample received starting from when you enable it until when you disable it.  That's 750,000 datapoints per second; something like a gigabyte per minute.

rp10007, 2017-09-17:
Thanks! Missed that one by a week.

I don't know if one of your goals is to make a subsystem for  more flexible and potentially less data-intensive capture, but if so that would be great. (In particular, capturing at a much lower rate, whether averaged or not, capturing for a specified time, or starting at some trigger condition[s]). If not, I'll just start in on the postprocessing...

EspoTek, 2017-11-09:
Latest update (on git only ATM) now asks you where you'd like to save your files.

EspoTek, 2017-11-29:
If you're on Windows, a binary is available (as per special request from another user).
Binaries for other platforms will be coming soon, once I've added the icon to the Mac version and implemented I2C decoding.

closed 2017-11-29 by EspoTek