espotek-org/Labrador #13: Connection issues

drtrigon, 2017-10-18:
I have issues connecting the labrador hardware and get the software running. Here is what I tried:
* kubuntu 14.04: the c/c++ libraries are to old
* kubuntu 16.04: crash
* windows 7: gets recognized as "labrador" in device manager until I start the software, then it changes to "ATmega..." - but was never able to connect

I have to mention that I had to test all this in VirtualBoxes but nevertheless it should be possible to get it running, right? I need the labrador desperately in order to work out some 1-wire timings. :)

EspoTek, 2017-10-18:
No idea if it's a VirtualBox issue, but it's possible.
Which one of those is the native OS?

The first time you connect, it will enter an auto-firmware update.
The process is:
 - Set the bootloader flag.
 - Reboot the device into "Atmel bootloader" mode (changes the device, as far as the OS is concerned).
 - Flash the new firmware.
 - Reboot the device once to clear the boot flag.
 - Wait a bit.
 - Reboot the device again to enter "Labrador" mode.

Sometimes the second reboot fails.  It all depends on specifics of the USB timings.
Fixing it is as simple as unplugging the board and putting it back in.

If not, try a manual flash.
https://docs.google.com/document/d/1R5jLZKFCEUMIUj2vA2UjlZvy4phVSa5Dmz0Yy7oBarc/edit#heading=h.l7n6121mw9ba

If you're on 64-bit Windows, you'll want to edit the .bat to flash labrafirm_0003_01.hex.
For 32-bit, you want labrafirm_0003_02.hex.

Does it say "device not connected" above the oscilloscope controls, by the way?

drtrigon, 2017-10-18:
Those were all virtualboxes. (I don't like testing on my native system.)

The only thing I did recognize of the "auto-firmware update" is that it seems it was rebooted in "Atmel bootloader" mode but then I assume the device did not appear fast enough (due to virtualbox) in order to continue with the process. Anything I can do here? (tried to add usb device filters in order to have them auto-connected by virtualbox - did not help)

Does the manual flash work with the same scheme of rebooting and thus changing the driver?

> Does it say "device not connected" above the oscilloscope controls, by the way?
Yes, this is what I am referring to when saying "was never able to connect".

drtrigon, 2017-10-18:
Got it working my manually flashing. Had to modify the batch a bit in order to have enough time to reconnect (virtually) the device before the last step:

> REM cd dfu prog
> dfu-programmer atxmega32a4u erase --force
> REM dfu-programmer atxmega32a4u flash labrafirm_0001_01.hex
> dfu-programmer atxmega32a4u flash labrafirm_0003_01.hex
> dfu-programmer atxmega32a4u launch
> REM timeout 2
> ECHO Please make sure the device is connected again before continuing!
> PAUSE
> dfu-programmer atxmega32a4u launch
> PAUSE

Now it works and I am able to connect! Cool! Thanks for your support!

However ... the interface is not very responsive, actually it's hard to use it at all. I guess part of this slow response is due to the virtualbox but may be there is something else I could tune to improve?

closed 2017-10-18 by drtrigon

EspoTek, 2017-10-18:
Glad it's working, and sorry for the hacky firmware flashing.  It was a last minute addition necessary to get the board working on Android.

Performance issues are probably due to VirtualBox, maybe the way it handles USB calls.

On a native boot, you should be able to get 60 frames per second with zero dropped packets on a Pentium 4, as long as you're not running serial decode.  (I haven't tested this specific setup, but I have disabled all but 1 core of a Phenom II 965 and clocked it down to 1.6GHz.)

drtrigon, 2018-02-21:
Addendum: No VM needed anymore due to https://github.com/EspoTek/Labrador/issues/22.