espotek-org/Labrador #15: Feature request: Frequency measurement

aalku, 2017-11-09:
It would be convenient to have a frequency statistic on oscyloscope inputs or something like that to measure frecuencies. I currently do it pausing and then using the cursor but it's hard to mark the same spot in two consecurive cycles. The software could do an average of several cycles or something. I don't know if it's very hard to compute or not.

EspoTek, 2017-11-09:
I don't think it would be easy to compute for arbitrary waveforms.   Can't think of a way to handle waveforms where a given voltage is crossed over twice.

When using the cursors, use the point where the waveform intersects one of the horizontal guidelines rather than maxima/minima.  This makes it much easier to get exactly the same spot in two consecutive cycles.

aalku, 2017-11-12:
> When using the cursors, use the point where the waveform intersects one of the horizontal guidelines rather than maxima/minima. This makes it much easier to get exactly the same spot in two consecutive cycles.

I thing doing it by code might be the same. Measure when the wave crosses a level and calculate the period then the frequency. Working with irregular waves might be a hell but we could focus on simple waves. The main problems are where to place the level and the noise. 

The noise might transform a level cross to three crosses very close together. I would consider the level has been crossed if the average of the last N samples crossed the level or something like that. Maybe when the wave is on one side of the level and then N consecutive samples are on the other. This is imprecise but we can compensate putting several cycles together. Count the time it took to cross the level M times on the same direction (or 2 * M on any direction) and then divide by M.

About the level the average of the last 2 seconds would be a place to put it, or somewhere between the max and the min of those seconds. 

To support compund waves and get the frequency of the strongest one you might place several levels and use the lowest detected frequency.

closed 2018-02-05 by EspoTek