espotek-org/Labrador #16: Logic analyzer decoding.

aalku, 2017-11-09:
I wish to have logic analyzer decoding improved but I don't know exactly how. Logic analyzer recording would be very useful and maybe we could decode it later. You might be recording a signal not knowing the baudrate or even the protocol. I would want to try several protocols on the recorded signal. Maybe having new protocols and configurations too. Yesterday I missed NEC Infrared protocol. I know it's too specific but if the decoding is structured in a certain way that it is easy to extend maybe the users will send pull requests with new protocols. An option to reverse the signal would be nice too.

aalku, 2017-11-09:
Maybe we could replay recorded signals too 😄 or even encoding the signal too 😉 I mean for example writting text as serial data signal.

EspoTek, 2017-11-09:
Yeah, the logic analyser definitely is lacking.  And it's mainly just because of a lack of software too.
I'll look at how to structure the decoder more easily so that users can write their own.
It won't work for all protocols, but ones that are "UART-like" should be possible to implement pretty easily.

And playback is definitely simple to do, at least for files smaller than the user's RAM.

EspoTek, 2017-11-29:
The latest version of the source code has the serial decoder implemented in a separate class.
Moving towards a "plugin" approach...

closed 2018-04-04 by EspoTek