espotek-org/Labrador #17: Simple icon

grocid, 2017-11-16:
Hi,

seemingly, there is no icon present for macOS so I made one.

[image]

Feel free to use it, there is no copyright issues.

All the best,
Carl

EspoTek, 2017-12-05:
Thanks, Carl.  
Unfortunately there's already an icon for other platforms, so it would probably best to use that one on Mac too.

If you feel like jazzing it up a bit, I'll upload it below.  If not, I can work something basic out in Paint.NET.

grocid, 2017-12-07:
[image]

Something like this?

EspoTek, 2017-12-07:
Yeah man, that's awesome!  Thanks so much.

ChrisWeiss, 2017-12-08:
Sweet! It needs a scope trace or something. Maybe make it old-school green
phosphor in color :)

EspoTek, 2017-12-30:
Check the releases page.  Icon has been implemented!

closed 2017-12-30 by EspoTek