espotek-org/Labrador #19: Device not connected on 3 different systems

pmpbar, 2017-12-04:
Hello! The desktop software is not picking up my Labrador in macOS, Debian linux and also Windows 8.1. I can see an Atmel Corperation usb vendor ID when I check but the oscilloscope software will not pick it up. The linux system was a virtualbox but the windows and Mac systems were real computers. Is there some trick to getting the little guy to get picked up? I tried switching around to all the ports on the machines to see if that would help. Let me know what info is needed to help.

EspoTek, 2017-12-05:
Sounds like something went wrong with the firmware updating process.

Plug in the board on Windows, navigate to the /firmware directory of your installation and run 'flash.bat'.

Faster and more stable firmware flashing is something that will be in the next few updates.  It's already implemented in Windows (source only ATM), just need to copy and paste a bit of code to get it running on Linux/Mac/Android.

EspoTek, 2017-12-05:
Also note that it will reflash the firmware whenever you switch between Windows x64 and other platforms.  This is to help overcome bandwidth limitations on Windows.

Make sure you don't unplug the board during this flashing stage (approx 10 seconds).

pmpbar, 2017-12-05:
I am getting a device not found when I run the flash script. Maybe my board is bunk?

EspoTek, 2017-12-05:
You said you got an Atmel VID...  What's the PID?

EspoTek, 2017-12-05:
Or, on Windows, what does device manager say?

pmpbar, 2017-12-08:
Sorry for the delayed response, the PID is ba94. `ID 03eb:ba94 Atmel Corp.`

EspoTek, 2017-12-08:
Yep, that's the correct VID/PID.  I suspect it's just clashing with another device.  Unplug all other devices connected to USB2.0 ports and connect Labrador through the 2.0 port.  If you connect it though a hub, ensure that it's the only device on the hub.  If you have a spare hub, connect  only the Labrador to the hub, and plug on through the hub.  That should fix it.  Long story short, it's a quirk of Atmel's USB controller when dealing with isochronous connections, mixed with the way that most USB devices handle 12mbps devices.

The troubleshooting section of the documentation has a short explanation and a few tips.

pmpbar, 2017-12-08:
Hmm, I have it directly connected. I will mess around with it more and see what I can learn.

EspoTek, 2017-12-09:
Interesting.  No keyboard or mouse on the bus?

Does the cable you're connecting it through support data as well as power?  Some cheap ones don't.  Try connecting a phone to the PC and see if you can transfer data.

EspoTek, 2017-12-09:
The red light will come on even if there's no data line, by the way.  It just means that the power supply is active.

EspoTek, 2017-12-20:
Did you end up getting it working?

closed 2018-02-05 by EspoTek

saudalwasly, 2021-09-29:
I have the same problem with Linux. I do not have a Windows machine. Is there any way to connect to the board?
What driver are you relying on?

EspoTek, 2021-09-30:
If you have Linux, could I please ask you to send me output of `lsusb` both when the board is disconnected and when it's connected?

No driver is required on Linux.

saudalwasly, 2021-09-30:
I have got it working now by pluging it in a USB2 port. For some reason it
doesn't work with USB3 port!

Is there any explanation?

EspoTek, 2021-09-30:
>Is there any explanation?

Yes, it was most likely bandwidth limitations:
https://github.com/EspoTek/Labrador/wiki/Troubleshooting-Guide

I'm guessing there was a webcam or other device plugged into one of your USB3 ports.

EricShalov, 2024-10-03:
Having the same problem, on MacOS 14.5 on a 2.6GHz MBP. I plug it in using a known-good not-only-power cable, it then appears on the bus when I run 'ioreg -p IOUSB', as this:
    	  | +-o IOUSBHostDevice@14200000  <class AppleUSBDevice, id 0x1008246b6, registered, matched, active, busy 0 (1 ms), retain 14>

..but when I run the Labrador MacOS/X app, it just says "Device Not Connected!"

EspoTek, 2024-10-04:
> Having the same problem, on MacOS 14.5 on a 2.6GHz MBP. I plug it in using a known-good not-only-power cable, it then appears on the bus when I run 'ioreg -p IOUSB', as this: | +-o [...]

What does `lsusb` show when the board is connected (you can install lsusb through homebrew)

taybart, 2024-10-18:
>What does lsusb show when the board is connected (you can install lsusb through homebrew)

On macos you can also use `system_profiler SPUSBDataType`