espotek-org/Labrador #21: Enable data storage in Android app

drtrigon, 2017-12-17:
Please add data recording/storage/logging to the Android app, would that be possible?

EspoTek, 2017-12-20:
Power supply and data storage functions are actually all in the software.  I just hide the buttons to activate them on Android.

The reason I did this is because of the small memory on most phones.  8GB free space or less is very common, and that would fill up with something like 10-15 mins of recording.

The power supply option is disabled since USB OTG is specced for 100mA only, and most of that would be eaten up just by the board itself.  I think I calculated 65mA with no signal generator or power supply load, leaving only 35mA for the power supply (which would be almost completely lost in the switching circuitry).

That being said, I plan on improving the DAQ features of Labrador, and once that's in then I'll be able to add the data storage back into Android.
I'll be working on it during the week after Christmas.

Enabling the power supply is not really an option, though.

ajxn, 2017-12-22:
Maybe give a short not about this in a About message dialog?

drtrigon, 2017-12-23:
> The reason I did this is because of the small memory on most phones. 8GB free space or less is 
> very common, and that would fill up with something like 10-15 mins of recording.

What about a screenshot feature; store the contents of the display - e.g. single trigger - as png or csv?

What about an expert feature that unhides the hidden features with a warning about the issues.

> The power supply option is disabled since USB OTG is specced for 100mA only, and most of that
> would be eaten up just by the board itself. I think I calculated 65mA with no signal generator or [...]

I assumed something like that... that's a pitty because it would be useful not as a current source but signal voltage source, e.g. for testing a ADC port (drawing microamps).

> That being said, I plan on improving the DAQ features of Labrador, and once that's in then I'll be 
> able to add the data storage back into Android.
> I'll be working on it during the week after Christmas.

Great news, perfect! Will be a very nice (Christmas) gift!

EspoTek, 2017-12-23:
The screenshot feature is also already in there.  It's hidden on Android too, but I can't offer a good explanation why...

I could enable an "Expert Mode" that turns on all of the features, and it could pop up with ajxn's "About" message that warns the user when you click it!

drtrigon, 2018-01-21:
Can you give an assumption when the new updated app will be available?

EspoTek, 2018-02-05:
When I get around to it.  ;)
Should be soon.

Have quit the PhD and so will be unemployed as soon as I finish a piece of software for the honours students.  We have a neurosignal amplifier only runs on a single PC that runs Windows XP and requires a DRM dongle.  I managed to reverse-engineer the protocol, but now that needs to be turned into an API.
Then, it's Labrador time.  :)

EspoTek, 2018-02-11:
It's live.
You should get a notification from Google Play soon.

closed 2018-02-11 by EspoTek