espotek-org/Labrador #22: Please provide an AppImage for Linux

probonopd, 2018-01-13:
Now that you are using `linuxdeployqt` you might consider to provide an AppImage for download, e.g., on GitHub Releases:

https://github.com/EspoTek/Labrador/blob/8ebd95e928198b75ce2fb5004e3363fc85650bcf/build_scripts/linux%20deploy%20labra%2064-bit/autodeploy#L28

`linuxdeployqt` has an `-appimage` switch that would make it easy.

- Applications packaged as an AppImage can run on many distributions (including Ubuntu, Fedora, openSUSE, CentOS, elementaryOS, Linux Mint, and others)
- One app = one file = super simple for users: just download one AppImage file, [make it executable](http://discourse.appimage.org/t/how-to-make-an-appimage-executable/80), and run
- No unpacking or installation necessary
- No root needed
- No system libraries changed
- Works out of the box, no installation of runtimes needed
- Optional desktop integration with `appimaged`
- Optional binary delta updates, e.g., for continuous builds (only download the binary diff) using AppImageUpdate
- Can optionally GPG2-sign your AppImages (inside the file)
- Works on Live ISOs
- Can use the same AppImages when dual-booting multiple distributions
- Can be listed in the [AppImageHub](https://appimage.github.io/apps) central directory of available AppImages
- Can double as a self-extracting compressed archive with the `--appimage-extract` parameter

[Here is an overview](https://appimage.github.io/apps) of projects that are already distributing upstream-provided, official AppImages.

If you have questions, AppImage developers are on #AppImage on irc.freenode.net.

drtrigon, 2018-02-04:
Any other kind of package that does not need to be installed would be great.

EspoTek, 2018-02-05:
Somehow missed the original post.
AppImage looks useful, but there's still the problem of udev rules needing to be installed.

Does AppImage offer any particular advantage over .deb in this instance, since the user needs to run an install script anyway (Apart from the Live ISO support)?

probonopd, 2018-02-09:
> AppImage looks useful, but there's still the problem of udev rules needing to be installed.

You can use a custom bash script to install them. For Arduino, I am using this as part of a custom `AppRun`/start script which appears to be working for me, at least on Live ISOs (more testing on installed systems needed):

```
cat > /tmp/roothelper <<\EOoF                                                                                                                                                           
#!/bin/bash                                                                                                                                                                             
cat > /etc/udev/rules.d/50-arduino.rules <<\EOF                                                                                                                                         
SUBSYSTEM=="tty", KERNEL=="ttyUSB[0-9]*|ttyACM[0-9]*", GROUP="users", MODE="0666"                                                                                                       
EOF                                                                                                                                                                                     
udevadm trigger
EOoF
chmod a+x /tmp/roothelper

sudo true && sudo /tmp/roothelper
sudo true || pkexec --disable-internal-agent /tmp/roothelper || true

rm /tmp/roothelper
```

Intended behavior:
* On Live ISOs, it just becomes root, does its thing, without asking the user
* On installed systems, it asks the user for the root password

probonopd, 2018-02-09:
> Does AppImage offer any particular advantage over .deb in this instance, since the user needs to run an install script anyway (Apart from the Live ISO support)?

Yes. No installation needed, the user needs to do nothing besides download, set the executable bit, run.

EspoTek, 2018-02-10:
Hmmm, nice.
Well if implementing the udev rules are that easy, I'm happy to provide an .AppImage.  :)

Thanks for linuxdeployqt, by the way.
Works a treat.  I don't know if you've already got a piece of hardware (I'm guessing you searched for "linuxdeployqt" on GitHub and found this repo?), but I'd be happy to send you a board.  :)

Email me at admin@espotek.com with your address.

EspoTek, 2018-02-10:
Tried to build it, but no luck.
linuxdeployqt simply won't read the icon referenced in the .desktop file, even when I put it in the exact directory as recommended by the software's output.  When I remove the "Icon=*" from the .desktop file, it complains that there is no icon and refuses to continue.

Any ideas, or should I just keep it as is?

probonopd, 2018-02-10:
I am looking into the icon issue.

probonopd, 2018-02-10:
Please check
* https://github.com/probonopd/Labrador/blob/patch-1/Desktop_Interface/build_linux/AppRun
* https://github.com/probonopd/Labrador/blob/patch-1/.travis.yml
* https://github.com/probonopd/Labrador/releases

Due to lack of a device I can't test the full functionality right now, but it should get you started.

EspoTek, 2018-02-13:
Just having a quick look before work.

It requests root when you run it.
[image]

After a while it crashes during the destruction of the USB thread.
I did manage to get it working once, though.

Seems to mirror Issue https://github.com/EspoTek/Labrador/issues/23

How did you make it?

probonopd, 2018-02-16:
> How did you make it?

This is how I made it:
https://github.com/probonopd/Labrador/blob/patch-1/.travis.yml

EspoTek, 2018-02-17:
I manually built using a slightly modified version of your travis.yml file, got a working .AppImage.
Still had the same issue with the crash.  
It seems to only happen in the .AppImage, and only when the board is connected to the PC before launching it (on the board reconnect).  

Had a look through the source code to see precisely what could be causing this, and honestly what's most surprising is that it doesn't crash when built as a standard executable.  Looks like some maintenance is required.

EspoTek, 2018-02-17:
Went over the code and fixed the bug causing the crash.
It was an old bit of code that put the USB handling thread in an infinite wait loop rather than terminating it properly.
Embarrassing, to say the least...

The latest version has fixed this, though.  Please submit your PR!

drtrigon, 2018-02-17:
Thanks a lot for providing the AppImage that's very nice!

Still an issues ... as I a bound to (k)ubuntu 14.04 I get following error:

> /tmp/.mount_EspoTelErDdL/usr/bin/Labrador: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /tmp/.mount_EspoTelErDdL/usr/bin/Labrador)

> /tmp/.mount_EspoTelErDdL/usr/bin/Labrador: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.20' not found (required by /tmp/.mount_EspoTelErDdL/usr/bin/Labrador)

A Possible solution that works for me is: https://askubuntu.com/questions/575505/glibcxx-3-4-20-not-found-how-to-fix-this-error
but as said I cannot do this as it is to intrusive.

Would it be possible to compile the AppImage against older versions of libstdc++6 also? That would be great!

EspoTek, 2018-02-17:
I also put up an AppImage in this repo, compiled using Qt 5.10 from my own system (Ubuntu 16.04).

I didn't check what libs it uses, so it's possible it requires v 3.4.21.

Simon, would you be able to sent through the PR so that drtigon can make an AppImage for his system?

probonopd, 2018-02-17:
I have sent the PR. It builds the AppImage on Ubuntu 14.04 on Travis CI so it should not have the issue @drtrigon described.

drtrigon, 2018-02-17:
No. I was using the AppImage from https://github.com/espotek/labrador/releases which gives me the issue. Your on the other hand runs like a charm - thanks a lot. What's the difference?

closed 2018-02-18 by EspoTek

EspoTek, 2018-02-18:
I've integrated probonopd's build system.
He's clearly better with his Linux-fu than I am.  :)

Check the "continuous" release on the page.  That should work on 14.04 and include the new patch that removes crashes on thread destruction.

drtrigon, 2018-02-20:
I can confirm that the 5df9da5 release works for me even if the labrador is connected before starting the software (so the patch works too). Very very nice! Thanks!

ps.: Now I can use it on my linux machine, no need for VMs anymore - just brilliant! ;) Next up; I'll test the new APK - got it through http://apkleecher.com/?id=espotek+labrador

EspoTek, 2018-02-20:
Nice.  Looks like you've made someone's day, Simon.  :)
Thanks once again!

commit 5efb953d (2018-02-18): Merge pull request #34 from probonopd/master

Generate AppImage on Travis CI for each git push, closes #22