espotek-org/Labrador #23: Crashes after PSU DUTY CYCLE out of range

jdrowell, 2018-02-01:
Hi I've got two boards, both exhibit the same behavior. Testing this on Arch Linux. I cloned the repo and compiled the app, and it actually seems to run one every hundred times or so, but 99% of the time I get this:

Feb 01 01:27:07 beast Labrador[22902]: CHECKING CONNECTION!
Feb 01 01:27:07 beast Labrador[22902]: Entering unixUsbDriver::usbInit
Feb 01 01:27:07 beast Labrador[22902]: Libusb context initialised
Feb 01 01:27:07 beast Labrador[22902]: Device found!!
Feb 01 01:27:07 beast Labrador[22902]: KERNEL DRIVER INACTIVE
Feb 01 01:27:07 beast Labrador[22902]: Interface claimed!
Feb 01 01:27:07 beast Labrador[22902]: Connected
Feb 01 01:27:08 beast Labrador[22902]: BOARD IS RUNNING FIRMWARE VERSION 0x0004
Feb 01 01:27:08 beast Labrador[22902]: EXPECTING FIRMWARE VERSION 0x0004
Feb 01 01:27:08 beast Labrador[22902]: FIRMWARE VARIANT = 0x02
Feb 01 01:27:08 beast Labrador[22902]: EXPECTED VARIANT = 0x02
Feb 01 01:27:08 beast Labrador[22902]: Connecting now!
Feb 01 01:27:08 beast Labrador[22902]: newDig
Feb 01 01:27:08 beast Labrador[22902]: isoCtx submitted successfully!
Feb 01 01:27:08 beast Labrador[22902]: [n, k] =  0 0
Feb 01 01:27:08 beast Labrador[22902]: t =  1517455628300
Feb 01 01:27:08 beast Labrador[22902]: Delay =  0
Feb 01 01:27:08 beast Labrador[22902]: Setup successful!
Feb 01 01:27:08 beast Labrador[22902]: MAIN THREAD ID 0x7fc741ddb800
Feb 01 01:27:08 beast Labrador[22902]: Iso Stack initialised!
Feb 01 01:27:08 beast Labrador[22902]: SUB THREAD ID 0x7fc71feb6700
Feb 01 01:27:08 beast Labrador[22902]: PSU DUTY CYCLE of dutyTemp = 16 OUT OF RANGE (could underflow on SOF)!!!  ABORTING!!!
Feb 01 01:27:08 beast Labrador[22902]: PSU DUTY CYCLE of dutyTemp = 11 OUT OF RANGE (could underflow on SOF)!!!  ABORTING!!!
Feb 01 01:27:08 beast Labrador[22902]: PSU DUTY CYCLE of dutyTemp = 6 OUT OF RANGE (could underflow on SOF)!!!  ABORTING!!!
Feb 01 01:27:08 beast Labrador[22902]: dt_userWantsToCalibrate 65536
Feb 01 01:27:08 beast Labrador[22902]: ReinitUsb Stage 1 complete
Feb 01 01:27:08 beast Labrador[22902]: Cancelled 1 0
Feb 01 01:27:08 beast Labrador[22902]: Total Cancelled = 1
Feb 01 01:27:08 beast Labrador[22902]: libusb_cancel_transfer FAILED
Feb 01 01:27:08 beast Labrador[22902]: ERROR LIBUSB_ERROR_NOT_FOUND
Feb 01 01:27:08 beast Labrador[22902]: Cancelled 2 0
Feb 01 01:27:08 beast Labrador[22902]: Total Cancelled = 2
Feb 01 01:27:08 beast Labrador[22902]: libusb_cancel_transfer FAILED
Feb 01 01:27:08 beast Labrador[22902]: ERROR LIBUSB_ERROR_NOT_FOUND
Feb 01 01:27:08 beast Labrador[22902]: Cancelled 3 0
Feb 01 01:27:08 beast Labrador[22902]: Total Cancelled = 3
Feb 01 01:27:08 beast Labrador[22902]: libusb_cancel_transfer FAILED
Feb 01 01:27:08 beast Labrador[22902]: ERROR LIBUSB_ERROR_NOT_FOUND
Feb 01 01:27:08 beast Labrador[22902]: PSU DUTY CYCLE of dutyTemp = 1 OUT OF RANGE (could underflow on SOF)!!!  ABORTING!!!
Feb 01 01:27:08 beast Labrador[22902]: Cancelled 0 0
Feb 01 01:27:08 beast Labrador[22902]: Total Cancelled = 4

So the board is detected, firmware is OK (I updated it in one board, the other was up to date), but I get those PSU DUTY CYCLE errors and after that the app crashes:

Feb 01 01:28:53 beast Labrador[25267]: Cancelled 2 0
Feb 01 01:28:53 beast Labrador[25267]: Total Cancelled = 4
Feb 01 01:28:53 beast Labrador[25267]: libusb_cancel_transfer FAILED
Feb 01 01:28:53 beast Labrador[25267]: ERROR LIBUSB_ERROR_NOT_FOUND
Feb 01 01:28:53 beast Labrador[25267]: PSU DUTY CYCLE of dutyTemp = 0 OUT OF RANGE (could underflow on SOF)!!!  ABORTING!!!
Feb 01 01:28:57 beast Labrador[25267]: ReinitUsb entering stage 2
Feb 01 01:28:57 beast Labrador[25267]:

                                       unixUsbDriver destructor ran!
Feb 01 01:28:57 beast Labrador[25267]: QThread: Destroyed while thread is still running
Feb 01 01:28:57 beast systemd[1]: Started Process Core Dump (PID 25367/UID 0).
Feb 01 01:28:57 beast systemd-coredump[25369]: Resource limits disable core dumping for process 25267 (Labrador).
Feb 01 01:28:57 beast systemd-coredump[25369]: Process 25267 (Labrador) of user 1000 dumped core.

If I start the app, wait a bit, and then connect the board, it doesn't crash (and shows the device as "connected"), but nothing seems to work (no yellow "signal" line on grid). I haven't managed to use the app yet due to this problem.

jdrowell, 2018-02-01:
Just got it to work once after a ton of tries, and when it doesn't crash I get the following output:

Feb 01 02:25:36 beast Labrador[25832]: MAIN THREAD ID 0x7f3cdf4e7800
Feb 01 02:25:36 beast Labrador[25832]: Iso Stack initialised!                                 
Feb 01 02:25:36 beast Labrador[25832]: New voltage = 4.5   
Feb 01 02:25:36 beast Labrador[25832]: Going to send value  31
Feb 01 02:25:36 beast Labrador[25832]: SUB THREAD ID 0x7f3cbd2d5700
Feb 01 02:25:36 beast Labrador[25832]: newDig
Feb 01 02:25:36 beast Labrador[25832]: Changed to mode 0
Feb 01 02:25:36 beast Labrador[25832]: newGain = 8.000000
Feb 01 02:25:36 beast Labrador[25832]: gainMask = c0c
Feb 01 02:25:36 beast Labrador[25832]: Multiplying by  0.0625
Feb 01 02:25:36 beast Labrador[25832]: Buffer shifted by -4
Feb 01 02:25:37 beast Labrador[25832]: unified debug is of size 27

I didn't look at the code yet, but it seems that the SUB THREAD is starting too early most of the time (before the voltage is set)?

EspoTek, 2018-02-01:
Voltage settings are communicated by the main thread, not the sub thread, so I don't think it's that.
dutyTemp should never be 1 unless there's something very, very wrong with the software.  I suspect that maybe there was some kind of compilation error.  Look in the .pro file and see what CFLAGS I use.  I have a feeling you need -O0.

Can you try the precompiled one and let me know how that works?

Regarding the yellow line, that's generally a symptom of limited USB bandwidth.
Hit help->firmware recovery and follow the prompts.

jdrowell, 2018-02-01:
Your prebuild binary does work reliably, just tested it. I look at the compiler flags in use and there doesn't seem to be any optimization flags set, and actually they seem to be stripped in case of the release version. Maybe this is a recently introduced bug? Which commit was the .deb generated from?

Thanks for your help. Now I can use boards :) Next step is generating working packages for Arch Linux.

EspoTek, 2018-02-01:
The .deb is pretty recent; 6248f5d IIRC.  I don't think there have been any changes to the Linux part of the code since the .deb was released.

Also, is there no way to install .deb packages on Arch?  Manually invoking dpkg -i, for example?

jdrowell, 2018-02-01:
You could get the .deb installed, but that would not be integrated with the native package manager (pacman). The Arch way is to introduce a package in the AUR (https://aur.archlinux.org/). That way anyone can install a user-generated package (if they accept the risk). If it becomes popular, it will eventually make it to the official packages. I'll give it a go and making a package for Labrador.

Thanks for your help, I'm enjoying using the board with the binary build until I get it to build from sources on my system.

EspoTek, 2018-02-05:
Good to hear.
Hopefully you can get it running from source.
The only recommendations I can give is to build through Qt Creator and ensure you're using Qt 5.7 or later.

But if not, enjoy the prebuild executable.  :)

closed 2018-02-05 by EspoTek

EspoTek, 2018-02-17:
Hey man.
Two things:
 - I've managed to find the bug you were talking about and it's fixed in the latest version.  I think.  Would appreciate if you could test.
 - A customer has submitted it to AUR!