espotek-org/Labrador #25: Userinterface Android missing functionality? Mac Interface hangs very often.

ElenaPower, 2018-02-09:
I tried to use my Mac with Labrador but I am having big problems. App freezes very often for some seconds and zoom in oscilloscope view often hangs. Ctrl-Cursor/Zoom does not work on Mac because it's a system function to magnify the screen. And many other user interface culprits. Nearly useless for me.

So I tried the android version but with this I have big problems too.  I have a Samsung Galaxy S8. 

The text and input fields in the user interface are extremely small. And I e.g. cannot activate channel 2 in oscilloscope view or activate trigger. Where are these options in the android version. Underneath the oscilloscope view I only have Signal Gen Chan1 and 2 view. The others from the desktop app are missing.

How can I zoom in/out the horizontal view? It only works for vertical view.

What is Android special old person mode? No change for me.

Sorry but the software/userinterface on Mac and Android is useless.

EspoTek, 2018-02-09:
Hi Elena.

Can you please send me a screenshot of the interface on Android?  The text fields are small, but I'm curious to see how small.  Old person mode should make them bigger but you may need to reset the software before it works.

To access the other devices, swipe left and right on the bottom half of the screen.  It doesn't have the smooth animations of a modern Android app but it does work.

To zoom in and out in horizontal view, you need to tick a box on the oscilloscope page.  I didn't have a pinch zoom system that could detect direction in place by launch, so I used that instead.  I will definitely be updating it, as well as adding documentation, as soon as I have the time.
Weirdly enough, you're the first person to give any sort of negative feedback about the Android app.  Which surprises me, honestly, because it's nowhere near the standard of the desktop version.

Regarding the Mac problems, it sounds like the device is disconnecting fairly regularly.  Are you able to run the app in the terminal and send me the output, noting when the freeze occurs?
I'll also need to know where you got the app from and what version of MacOS you are running, in addition to what hardware you use.  
Other people are using the Mac version regularly just fine, so I'm not sure why yours in particular is failing.
Regarding the shortcuts, try pressing command instead of Ctrl.  I know this was mentioned somewhere in the documentation, but I think it would be best to update it with special instructions for the different platforms right at the top.

Hope that helps somewhat.  Get back to me with the rest of the information and we can hopefully work out the rest.
-Chris

EspoTek, 2018-02-11:
I have fixed the scaling on the Android app.  It now chooses between two presets; one optimised for 720 and one for 1080.  Even the 720 is much bigger than before.

It should be usable at least, now.

Please get back to me with a copy of the Mac App from terminal so I can help debug it with you.

Thanks,
~Chris

drtrigon, 2018-02-21:
> Weirdly enough, you're the first person to give any sort of negative feedback about the Android app. 
> Which surprises me, honestly, because it's nowhere near the standard of the desktop version.

;)) I like that statement and have to jump in. Honestly I was (positively) surprised that the Android app worked at all given e.g. the firmware flashing issues I had in linux https://github.com/EspoTek/Labrador/issues/13 (and my life experience of course ;). So I hardly tried to gave feedback on the main issues that made the app unusable. At its current state now, given https://github.com/EspoTek/Labrador/issues/21, https://github.com/EspoTek/Labrador/issues/36 and https://github.com/EspoTek/Labrador/issues/35 the only thing left that I could complain about is the mentioned troubles in switching left/right having to try 3-5 times often - but this does not make it unusable - it just makes usage a bit annoying.

EspoTek, 2018-02-21:
Haha, same.  I was not expecting something written in Qt to work with such minimal effort.  Of course, it looks like something from the Android 2 era, but at least now it's reasonable.

The trick with the swiping is to move fast and make sure you're not over a text box or anything.  I usually swipe over the labels and don't have any trouble.

EspoTek, 2018-02-21:
Mind you, that's still a pretty poor response.  Hence not closing the issue.

closed 2018-04-04 by EspoTek