espotek-org/Labrador #30: Hardware licensing

kasbah, 2018-02-10:
Hey, I was just watching the [recording of the FOSDEM talk](https://fosdem.org/2018/schedule/event/cad_spice/) and have been thinking about a PCB ruler/keychain to promote [kitspace.org](https://kitspace.org). I want to make something that could just be a nice reference of package sizes if not populated but be even more useful if populated. Seems like the Labrador would be an awesome design to try and do that with. 

Regarding the license, I am not sure how my use fits with the non-commercial aspect. I haven't made any money of Kitspace yet but I hope to start supporting development of it with affiliate links that I added recently (Aisler and PCBway). 

>The intention is to move to a less restrictive licence in future, where commercial hardware derivatives are allowed but the use of the Labrador name or any branding is not.

INAL but I believe you automatically get a trademark on the name and branding by selling things under that name. So all you really have to do is switch to CC-BY-SA but make clear that your brand and logo are not covered by creative commons so have to be removed in derivatives.

EspoTek, 2018-02-10:
Good to see the recording's finally up.  Curious to see it, and will have to thank Guillherme too. 

Are you able to explain what parts of the Labrador hardware you are using, and in what way?   Are you just planning to make keychains out of its unpopulated header, or keychain versions of a fully working device?  And will you be selling them?

kasbah, 2018-02-10:
I was thinking about integrating the entire design into a keychain format, adding extra components that will always be DNP (for package references). Adding AWG holes, trace-widths, ruler units in mm and inch on the side and possibly some other reference items. Something like [this Adafruit one](https://www.adafruit.com/product/1554#description-anchor) but smaller. 

I probably won't sell them but give away the PCBs. People that want to populate them could then buy the components by clicking on the links on Kitspace.org (I don't make any money on those links). 

Regarding trademarks, similar to copyright, I believe in most countries you don't need to register a trademark to own it and enforce it.

EspoTek, 2018-02-10:
Sounds cool.
Feel free to go ahead with your project.  You can sell the keychains or use affiliate links on the component purchase links too, if you want.

Only two things I ask are that you provide credit in the form of some text on the  board itself (e.g. "PCB design based on https://gitub.com/espotek/labrador") and also send a couple to me when they're ready.  I'd be really interested to see what they look like.  :)

You can use the logo if you want but just don't pretend you're the "real thing".

-Chris

kasbah, 2018-02-10:
Thanks! I'll let you know how I get on with it. Feel free to close this issue or leave it open as a reminder if you want to look into changing your current licensing.

EspoTek, 2018-02-10:
Done!

closed 2018-02-10 by EspoTek