espotek-org/Labrador #33: PRIORITY ISSUE: Still no software support for I2C and SPI

EspoTek, 2018-02-11:


EspoTek, 2018-08-07:
(To anyone reading the issues here, I've been actively working on this since early August.  Should be in soon!)

prograMAnimus, 2018-08-07:
I'd second that!
This feature would be immensely helpful for me, as I'm currently deliberating of using the Labrador-Board for education purposes and want my students to fiddle around with SPI/I2C in later practices.

EspoTek, 2018-08-12:
Cool.  Should be done within a week or two.  I plan on starting actual, live testing later this week.

urish, 2018-10-02:
Hi @EspoTek, can you please update on the status of this? 

Also, any chance for a new Windows release with these protocols?

EspoTek, 2018-10-02:
Hey Uri.  The short, honest answer is that I ran into some bugs that delayed the feature, then ran out of time before I could fix them.  Have recently moved house (4hr/day commute => 1.5hr), so should have some spare time in the near future.

urish, 2018-10-02:
Thanks for the update!

EspoTek, 2018-11-04:
Quick update:
Working through this issue right now.  Looks like the software routines for detecting start and stop conditions are correct now, but the stream coming from the device is slightly out of phase (by 1/375,000th of a second).  Since that happens to be the exact time it takes to read a single byte on the logic analyzer, I suspect this is due to differences in buffering (single vs. double) between the channels.

Tools to open my AVR project are about 1GB and our new place doesn't have a phone line, so might need to take the laptop down to the library.  Looks like I2C support will be in within 48 hours, though.

EspoTek, 2018-11-06:
Looks like it's mostly implemented (ended up finding a laundromat with free wifi, of all places), but I'm unable to test it at the moment as I can only find the one I2C device.  Closing this issue for now, and opening a new one: "I2C untested"

closed 2018-11-06 by EspoTek

EspoTek, 2023-01-19:
It isn't, that feature was dropped.