espotek-org/Labrador #35: Please put an Apk for Android on the release page

drtrigon, 2018-02-17:
Thanks a lot for https://github.com/EspoTek/Labrador/issues/21! I wanted to ask whether it would be possible to offer the apk file for download on https://github.com/espotek/labrador/releases also? I would like not use google play and all other download pages lag quite some time behind.

drtrigon, 2018-02-20:
The first place I was able to get it from was http://apkleecher.com/?id=espotek+labrador

EspoTek, 2018-02-20:
I can manage hosting an APK here.  I'll look into F-Droid as well.

closed 2018-03-06 by EspoTek

maxbla, 2019-07-03:
What happened to F-Droid? Would you accept a PR adding F-Droid support?

EspoTek, 2019-07-03:
Fell into the "too hard" basket.
You're welcome to submit a PR adding it to F-Droid, but do note that Android builds have been broken for some time.  Fixing it should be as simple as either #ifdefing out chunks of `mainwindow.cpp` or adding the element to the Android `mainwindow.ui`.