espotek-org/Labrador #36: Bug: DAQ settings and rotating device in Android app

drtrigon, 2018-02-20:
So I updated and tested the new android app:

Positive:
- a lot better readability(!)
- DAQ features requested in https://github.com/EspoTek/Labrador/issues/21 are present (though have not tested them yet)

Bugs to report:
- crash/hangs in DAQ settings dialog when saving
- rotating the android device (to landscape) does not look useful as on the right-hand side only shows "device not detected" - after rotating back this text is shown
at bottom (instead of top) and switching left/right (to sig. gen., multimeter, etc.) does not work anymore

EspoTek, 2018-02-20:
I don't get any crashes in the DAQ settings dialog, although it seems like there's no way to close the dialog except by pushing the "back" button on the phone.

Horizontal rotation doing that isn't a bug; it's a feature.  :P
When the device is connected correctly, the scope will become full-screen.
Regarding the "device not connected" reappearing at the bottom, that is a bug not really a priority.  I don't have any trouble switching left/right after rotating back.

This is on a OnePlus 2 and Galaxy Note 2 N7105, both running LineageOS @ 7.1.
Can you send through your OS version and device model?

drtrigon, 2018-02-21:
> I don't get any crashes in the DAQ settings dialog, although it seems like there's no way to close the > dialog except by pushing the "back" button on the phone.

Yes correct - the word "crash" is wrong it hangs with the dialog still open. However pushing the back button works for me too, so that's ok I can live with that work-a-round (now that it's documented here ;).

> Horizontal rotation doing that isn't a bug; it's a feature. :P
> When the device is connected correctly, the scope will become full-screen.

Ok I'll have to check that - sounds great.

> Regarding the "device not connected" reappearing at the bottom, that is a bug not really a priority. I > don't have any trouble switching left/right after rotating back.

I tested it again - you are right switch left/right does work after rotation. I did not notice because it does not work very well; usually I have to try 3-5 times in order to get it switching and must have done it wrong while testing ("placebo effect").

Regarding the "device not connected" reappearing at the bottom, actually this does not matter for me (I can live with that ;). Just mentioned it in order to describe the symptoms.

> This is on a OnePlus 2 and Galaxy Note 2 N7105, both running LineageOS @ 7.1.
> Can you send through your OS version and device model?

It a Fairphone 2 (still) with original Android 6.0.1.
(LineageOS... was also considering to try it... but somehow feared the trouble... ;)

closed 2018-02-21 by drtrigon