espotek-org/Labrador #41: Device not connecting android 7.0

JennyEverywhere, 2018-04-26:
I'd tried the release on github, and even with an OTG cable that powers the board it simply will not connect. Is there a secret handshake or config I need to tickle to make it look for the board, or is android 7 still not working yet?

EspoTek, 2018-04-26:
Hey Gwen.  Saw your email this morning too.
For others having the same problem:
She's on a Huawei Honor 8 running Android 7.0.  Using an OTG cable, it powers the Labrador but won't connect otherwise.  The red "device not connected" message is displayed even when the board connects.

Labrador should work on Android 7.  You shouldn't need to do anything special to get it to work (the software should reconfigure it automatically), although you may need to unplug and reconnect the board manually after firmware flashing on some devices.

First thing's first, does your board communicate with a PC, using the same MicroUSB cable as the one you're trying with the phone?  Some MicroUSB cables do not have data wires, and this has caused similar problems to the one you're describing now.

Secondly, can you connect other USB devices to your phone using your OTG cable?  That would rule out the possibility of an incompatible OTG cable.  (I have this problem with my OnePlus 2, which is physically type-c but electronically USB2.0.  No problems with power, but data just doesn't want to transfer).

Finally, where did you get the software from?  Google Play or elsewhere?  The version on Google Play has the fix for 7.0 applied.  If you don't have Google Play on your phone, I can forward you the latest APK.

Let me know if any of that solves the problem.
~Chris

JennyEverywhere, 2018-04-26:
Thanks for getting back to me, Chris.

I get exactly the same issue with the red "device not connected". I got the
app directly from Google Play, but have also tried the .apk listed as
Update 2 on the Github repository. No difference.

The OTG cable I got has micro-B on one end and usb-C on the other. I don't
think I have anything else TO try with it. Here's where I got the cable,
which claims power and sync capability:
https://www.amazon.com/gp/product/B01LONQ7R6/

I'll try connecting to the Labrador from my PC...after much fiddling, I
managed to get it working on my PC. I had to do the firmware recovery
several times, even deleting the driver and reinstalling it. To calibrate,
I had to connect the usb cable to a port on the back of the machine...might
be 3.0 or eSATA, not sure. But it works. Calibrating on my hub, there was
way too much noise and it wouldn't settle down close enough to zero.

Even with the firmware reflashed, it isn't coming up on the Android device,
though.

EspoTek, 2018-04-26:
Sorry about the issues with the PC.  I tried to make it compatible with as many systems as possible, but due to the way it synchronises transfers it can request huge amounts of USB bandwidth on modern systems and clash with keyboards/mice.

I checked out the cable and it looks like a fair few users are claiming it's not OTG compatible.

I'm not sure about the specifics of Type-C/3.1, but on old micro-USB2.0 connectors, there was an extra pin that would need to be shorted to ground to signify that it was for use in an OTG host.  When I connect Labrador to my phone, it has to be in the configuration Phone->USB-OTG adapter->MicroUSB Cable->Labrador.  It doesn't work the other way, even though both ends are MicroUSB, because the host pin is only shorted on the OTG adapter.

I have no idea why usb.org made this decision, but it makes about as much sense as their choices for the software stack too.  A classic case of design by committee.

JennyEverywhere, 2018-04-26:
I've got some other adapters and such on order so I can try some other configurations. It's all a bloody mess.

EspoTek, 2018-04-26:
Yeah, tell me about it...

Best bet would be an adapter that adds an old-school female type-A port to the phone, and then connecting a standard microUSB cable to that.

JennyEverywhere, 2018-04-26:
Exactly what's arriving on Friday. I'll let you know if it changes anything.

EspoTek, 2018-04-26:
Fingers crossed.  :)

EspoTek, 2018-05-21:
Does the new software fix the issues from before?

JennyEverywhere, 2018-06-15:
> Not on Linux. I'll try on Android soon.

Chris,
Just wanted to touch base. I gave up trying to run Linux on my workbench. I
also gave up on my old CRO, and trying to use USB scopes for anything but
learning tools. I got a Rigol 4-channel 50mhz scope which upgraded to
100mhz with a code, and a lovely little ASUS VivoMini running Win10. Works
PERFECTLY with Labrador, my Digilent AD2. I still wish I could make it work
with my Huawei phone, but that's probably the Huawei's fault, not
supporting some piece of OTG that Labrador needs to work right.

I still have one of my Labradors (I bought one, my partner bought 4)
plugged into a breadboard on my bench. I have a Toaster breadboard power
supply on another breadboard. And I have a third breadboard on a lovely
Lucite base, next to an Arduino Uno that I got on Adafruit.

I built a project that emulates an old Altair 8800 microcomputer, one of
the first commercially-available personal computers. It emulates the system
in software, running on an Arduino Due, with a custom front panel that
replicates the original main panel of the Altair. It's called an
Altair-duino.

https://www.altairduino.com/

Right now, I'm in a phase of building radios. I built a 40-meter CW
transceiver that used a DDS module and an Arduino Nano, and now I'm
building an all-band rig that's similar, but it's both CW and sideband
called the uBitX (micro-Bit-X) that uses a si5351 with a Nano to make a
"Raduino" module.

Here's the site for the second, all-band radio: http://ubitx.net/

I attached a copy of the original QST magazine article on the first CW
transceiver. It's below.

Thanks for all your help. If you want me to test anything on the Huawei,
just ask and I'll be glad to help you in return.

-- Gwen

EspoTek, 2018-07-19:
Hey Gwen.  Good to hear from you, and glad it's finally working.  And yeah, no hard feelings.  There are some things that a $500 Rigol product can do that Labrador can't.

Just had a look at the Altairduino and it seems pretty cool.  You thought about contacting Crowd Supply about it?  I'm sure they'd be happy to help you with it.

closed 2018-07-19 by EspoTek

JennyEverywhere, 2018-07-19:
The Altair-duino is already in kit firm from its creator. I just built one,
I didn't design it. 😉

EspoTek, 2018-07-24:
Hahaha, whoops.  Man, I'm tired.