espotek-org/Labrador #46: Window is not resizable.

alxpettit, 2018-08-06:
Window does not resize. This may not seem like a big deal, but as someone who uses a ultraportable laptop with a screen smaller than the size of the window, it's a _serious_ problem.

[image]

Please fix this.

alxpettit, 2018-08-06:
[image]

This is pretty much one of only two apps I can think of that actually won't work with 768 vertical pixels at standard DPMM. I don't think it's too much to ask? Let me know your thoughts.

EspoTek, 2018-08-07:
Hey Alex.  I got a seat.  ;)

Have pushed a patch to GitHub that fixes the rendering on low-res displays - down to a minimum of 800x600!  It was originally designed to run at a minimum of 1024x768 before I broke it, so I guess that's an imprevement.  :)

The build server should make it within the next 30 mins or so, and when it's ready it'll be available here:
https://github.com/EspoTek/Labrador/releases

You'll want the AppImage from the "Continuous Release" section.

If you want to install it, you should be able to build from source (qmake, make, make install).
I tested it on my machine running Xorg/i3wm and it worked a treat.

I'll mark this issue as resolved for now.  If it doesn't work on your machine, just ~~install a new DE~~  reopen it and I'll see what I can do.

EspoTek, 2018-08-07:
(And by the way, what was the other app that was broken on your machine)

alxpettit, 2018-08-07:
Thanks! ^.^ The other software is [this](https://sourceforge.net/projects/aiboplus/). I guess they're maintaining a pretty massive build chain and sizable codebase and didn't really focus on the GUI much at all. It actually even went off the screen on my mother's much bigger laptop. I'll have to contact them about that and let them know at some point. :stuck_out_tongue_closed_eyes:

alxpettit, 2018-08-07:
Hmm. Problem still seems to exist as of commit ddf60d5c63019eb27915d5d6a396d1374da071ce.

alxpettit, 2018-08-07:
Tried compiling from source, and I got the same issue... (by the way, AppRun didn't let me run it without installing very willingly. I had to edit line 20 to `exec "$HERE/../bin/Labrador" "$@"` and edit my `LD_LIBRARY_PATH` to point to the directory containing `libdfuprog-0.9.so` to get it working. I'll probably open an issue about that as soon as I figure out exactly what went wrong and whether I was using anything in an unintended way...)

Anyway, the issue still existed after compiling from source. Which isn't surprising. Honestly, I think compiling it from source was more of me being bored than an actual debugging step.

EspoTek, 2018-08-07:
Hmm, I'm not sure exactly what could be causing this.  I'm on a fairly stock version of Arch (also Xorg, but i3wm as window manager), and on my machine it renders at 800x600 without any issues.  I'm not sure of the specifics of DPI scaling either, but manually overriding the geometry should manually override it...

I'm about to go to sleep, but maybe try changing line 163 of MainWindow.cpp to `this->setGeometry(0, 0, 640, 480);`  Maybe even `this->setFixedSize(1366, 700)'?  
If you've got some weird scaling setting that might do it?
Possibly change your scaling settings, or full-screen it and un-fullscreen it again?

I'm not exactly sure what's causing this issue ; sorry.  
Either way, it should not be rendering at 1024x768.  That patch changed the minimum resolution down to 800x600.

EspoTek, 2018-08-07:
Actually, just thought of a possible issue...
Pull the latest version.  It *may* contain a fix.

alxpettit, 2018-08-07:
Just pulled and recompiled. No dice. :(

I'll go ahead and check the latest binary build, in case my `make clean` call left some old objects somewhere or something.

alxpettit, 2018-08-08:
If it helps with debugging (seems like our best bet at this stage), I could go ahead and try to recreate the issue in a virtual machine with a minimal standard Arch setup, and maybe with a few other distros. Once I get something with the problem, I could even send you the whole virtual machine file so you can do diagnostics. Would that help?

EspoTek, 2018-08-08:
I think it might be easier to just try it in a different desktop environment.  ctrl-alt-F2 will let you log into a second virtual terminal.

alxpettit, 2018-08-08:
Eh, it could be almost anything at this point, down to Xorg configs or bugs in specific package versions. Plus, I like to keep my setup clean. Installing packages not natively needed except for one test would leave lots of crud around my system to clean up. :stuck_out_tongue:

EspoTek, 2018-08-08:
Or perhaps adding a 68 pixel tall widget to the bottom of the Main Window would at least let you see the whole screen?  Should be easy to do graphically in Qt Creator.  Make sure it's fixed-size.  And ignore the mess...

alxpettit, 2018-08-08:
I just recreated the bug in a Xubuntu VM. Definitely not a fluke.

alxpettit, 2018-08-08:
Lubuntu won't display properly on my machine in a VM. I vaguely remember having this issue before. Not sure what to do about it, but unfortunately I won't be able to use it to test your build. :(

alxpettit, 2018-08-08:
I've got some bad news...

[image]

Um... now what?

alxpettit, 2018-08-08:
I'm doing some qt API research. I'll let you know if I find anything!

alxpettit, 2018-08-08:
Yep. I fixed your problem.

alxpettit, 2018-08-08:
Actually, it's such a small change that it's not needed. Change lines 18 and 19 in `mainwindow.ui` to correct minimum width.

I may have a few other pull requests though! If you don't mind that is.

EspoTek, 2018-08-08:
Oh, would love a PR.  What did you do to get it working in the end?  Just changed the .ui?  It's weird; setGeometry should have overwritten that setting as it was called after the UI initialisation.

alxpettit, 2018-08-08:
Yeah. Changing the `.ui` is all I needed to do.

EspoTek, 2018-08-08:
Too easy.  I can patch that tonight.

EspoTek, 2018-08-12:
3 days later...

Finall had the opportunity to sit down and fix the issue.
I'll mark this closed for now.  Feel free to pass on your other ideas.  I'd love to hear them.  :)

Cheers,
~Chris

closed 2018-08-12 by EspoTek