espotek-org/Labrador #47: Board not getting recognized by OS

prograMAnimus, 2018-08-08:
So I've tested this with 3 different Micro-USB cables, two different PCs with each running (natively) Windows10 and Linux Ubuntu + Arch.

Whenever I connect my Labrador-Board, I'd expect that the Labrador-Board would share it's usb identification with the OS. Neither _dmesg_ nor _lsusb_ show any indication of plugging in anything new - as if the Atmel-Chip never establishes the usb-connection.

Did I get a faulty firmware? How can I further debug the cause of this problem?

EspoTek, 2018-08-08:
It should show up in lsusb.  The fact that it isn't indicates a problem.

The simplest way to check if it's hardware or firmware is to try connecting it in Bootloader  mode.  There's instructions on the wiki showing how to do this.  Once in bootloader mode, you should be able to manually recover the firmware with dfu-programmer.  There's an example .bat file in the /firmware directory for Windows, as well as the dfu-programmer binary.  You should be able to just run that.

Aldo, just to double-check, have you tested your cable with an android phone and confirmed they carry data?  Its unlikely, but it could be a case of all 3 cables being power-only.

prograMAnimus, 2018-08-10:
In the meantime I've borrowed my colleagues' Labrador-Board - which was brand new as well btw - and connected it to my PC using the same USB-Cable I was using before.
Lo and behold: Dmesg is telling me that the board has connected and the desktop-application also connects to the board and displays the readings.

So we can at least take the cable out of the equation here.

Next I hooked up Dig. Out 1 to Gnd in order to switch to the Bootloader-Mode but that didn't do anything at all; in order to flash something, at least a connection must be established, but it still didn't.

Is there any other way to debug this or should I just give up at this point?

EspoTek, 2018-08-10:
Where did you buy the board from?  I'd recommend taking it back for a refund.  It sounds like a faulty board.

prograMAnimus, 2018-08-10:
Yeah I'll most likely do that, thanks for your timely answers anyway! :+1:

closed 2018-08-10 by prograMAnimus