espotek-org/Labrador #49: Add display scaling settings to main window

drtrigon, 2018-09-02:
It would be very useful if both graph axes ("Time Window" and "Vmax"/"Vmin") could be easily changed in the main window. Simplest would be to just move everything from "Oscilloscope > Range > Enter Manually" to the main window. May be also add sliders or something (allowing for zooming out also).

EspoTek, 2018-09-04:
Hey man.  Great to hear from you.  It's nice to see someone getting so much use out of the project!

I added this feature on the way home from work.  Check the latest build.  :)
Oscilloscope->Show Range Dialog on Main Page.

closed 2018-09-04 by EspoTek

drtrigon, 2018-09-04:
That was fast! Thanks a lot!

> Hey man. Great to hear from you. It's nice to see someone getting so much use out of the project!

It is a cool project! Thanks also for your ongoing and fast support!

EspoTek, 2018-09-04:
Haha, gotta make an effort for Dr. Tigon.  :)

drtrigon, 2018-09-11:
Work very nice and is extremely useful, thanks!