espotek-org/Labrador #50: Allow to select Oscilloscope CH2 only

drtrigon, 2018-09-02:
Currently it is not possible to switch off CH1 if this is needed e.g. to examine the signal from CH2 in more detail.

EspoTek, 2018-09-04:
Check the latest build.

I wasn't able to do a "true" CH2-only mode as that would require a firmware rewrite (yes, seriously!) but it can hide any graph when paused.

closed 2018-09-04 by EspoTek

drtrigon, 2018-09-10:
It would be nice to be able to disable Ch1 only but at the same time enable a digital channel. However this is already helpful thanks!

drtrigon, 2018-09-10:
Would it be possible to hide channels that are NOT paused?

drtrigon, 2018-09-10:
Hiding is done in the GUI?

EspoTek, 2018-09-10:
Yep, that's correct.  I don't want to have separate buttons for "enable CH1" and "hide CH1", though; it'll just cause confusion.  Perhaps I could move it up into the menus as a hidden-away option.  "Do not draw CH1 waveform" or similar.

drtrigon, 2018-09-11:
What about this: if I enable Ch1 and Ch2 and then disable Ch1 it just hides it (but still processes the data in background)? This behaviour can be enabled in the menu as you suggest. No hide button or option would be needed at all. Only drawback I can see is that no digital Ch will become available.

EspoTek, 2018-09-11:
Hmm, that is a little bit odd but it does make sense if you take notice of the buffer indicator in the corner.

Definitely better than my idea.  I'll try and get  it in this week.  :)

drtrigon, 2018-09-11:
That would be brilliant - thanks!

I think it makes sense from the user point of view - it behaves more like a desktop scope a user might already know. It might not make sense from the technical point of view. As you mentioned thanks to the buffer indicator a user interested in the technical details can still get insight and understand the exact behaviour. (Speaking for me only; sometimes I like getting more technical insight, but sometimes I just don't care at all, as I need to be able to get a job done as fast and easy as possible... ;)

Going the route of thinking as a user further down I think it makes sense to implement the same behaviour for the digital channels too.

So as I understand the technical part; clicking on Ch1 if Ch2 is active should not disable both but just hide Ch1 (keeping both buffers busy). As soon as I disable Ch2 too both buffers are free (so if Ch1 is hidden it does get disabled and may be unhidden). If I instead re-enable Ch1 (unhide) and then disable Ch2, 1 buffer gets free. And that should hold for analog and digital channels. Does that make sense?