espotek-org/Labrador #51: Signal syncronization when pausing

drtrigon, 2018-09-02:
I have the strong impression that when I pause the display of data or configure the trigger for single shot, the two signals from Oscilloscope CH1 and CH2 are not in sync. If I pause 10 times, about 9 times they are shifted by some time against each other. (Also the trigger seams not to be precise.)

EspoTek, 2018-09-05:
I am unable to reproduce this bug, and I can't think of a technical reason why it would exist.  Both CH1 and CH2 should come in the same packet, and neither channel should be updated except when the other one also is.

Are you able to reproduce this with CH1 and CH2 directly shorted to one another?  Perhaps it's a race condition.

drtrigon, 2018-09-10:
I did not short them but I only connected Ch1 and what you see on Ch2 is cross-talk, here an example of today:
[image]
[image]
... as you can see the spikes move with respect to the actual signal by an arbitrary amount of time. This was done using a trigger on Ch1. Actually when taking a closer look I think it is Ch1 that does get shifted - may it be related to using the trigger on that Ch?

EspoTek, 2018-09-16:
Whoops, this looks suspect:

```
    readData375_CH1 = internalBuffer375_CH1->readBuffer(window,GRAPH_SAMPLES,CH1_mode==2, delay + ((triggerEnabled&&!paused_CH1) ? triggerDelay + window/2 : 0));
    if(CH2_mode) readData375_CH2 = internalBuffer375_CH2->readBuffer(window,GRAPH_SAMPLES,CH2_mode==2, delay + (triggerEnabled ? triggerDelay + window/2 : 0));

```
Note the "TriggerEnabled" condition is different between the two channels!
I'll get a proper fix up shortly.

EspoTek, 2018-09-16:
I have just pushed a fix to this but am unable to test it right at the moment.  Are you able to confirm that it now works?

drtrigon, 2018-09-16:
Thanks! Yes I can confirm that both channels stay in sync now! When pausing both are moved now by the same amount of time. Nice would be if the don't move at all.

Is it possible that the trigger for "Ch1 (rising)" triggers the falling slope and vice-versa?

Also "AC Coupled" for Ch1 does not work - when I click it just pauses the measurement - interestingly in a correct manner (the channels are NOT moved at all and stay in sync). In order that AC coupling works I have to first click it on Ch2 and then on Ch1.

EspoTek, 2018-09-17:
Closing now.  Thanks again, Tigon.  :)

Regarding the AC Coupling, please open a separate issue with instructions to reproduce the bug.  I can't see what you see.

closed 2018-09-17 by EspoTek

drtrigon, 2018-09-17:
Done in https://github.com/EspoTek/Labrador/issues/55