espotek-org/Labrador #53: Segfault

alxpettit, 2018-09-10:
Running on: `Linux lithium 4.14.68-1-lts #1 SMP Wed Sep 5 14:38:31 CEST 2018 x86_64 GNU/Linux`

Labrador loads up, GUI looking fine, then [segfaults immediately](https://gist.github.com/alxpettit/9ef319e8a8893e9dcff95586271e47a1) after asking about calibration.

Compiled from source via:
```qmake
make
sudo make install
```
Compilation showed no anomalies.

EspoTek, 2018-09-10:
Which commit is this for?  I'm aware of a segfault in a recent dev build but it doesn't look like I pushed it.

alxpettit, 2018-09-10:
@EspoTek 7a2f8cf

EspoTek, 2018-09-10:
Had a closer look at what was going on.

I think the issue was introduced in this build:
https://github.com/EspoTek/Labrador/commit/2ba556bbe9112d82f9e3138caca50fd742ebb82b

It aborts when iso fails but the parent object does not acknowledge the failure.  This leaves midbufferoffets[1..3] uninitialised and when the first packet it received it tries to memcpy several thousand bytes somewhere random into memory.  At least that's what the output suggests.

Try reverting the commit I linked above and it should magically start working.  :)

Also, thanks so much for the bug report.  I thought the problem I had with the dev commit was related to the dialogue Window, but your report shows that this is definitely not the case.  :)

Don't have my laptop on me at the moment but I'll try and have a fix up soon.

Thanks again.  :)

alxpettit, 2018-09-10:
Aww, no problem. I like supporting FOSS and good hacking tools! ^.^

alxpettit, 2018-09-10:
Oh. I was browsing through the git log, and that's actually the one I had already checked out to build my install. I knew it would work because the one thing 2c2db8855bc66336f02a4509d6ae2069f1944fa7 did was fix the resolution scaling stuff from the issue I opened before!

alxpettit, 2018-09-10:
I'll go ahead and compile 2ba556b and see if the issue reappears, just to confirm.

alxpettit, 2018-09-10:
Uuuuuh. I compiled it and it's still working. Hmm.

EspoTek, 2018-09-10:
Oh, sorry.  That particular build only segfaults if iso init fails (ie, you have enough bandwidth to connect to the device but not enough to communicate with the oscilloscope's endpoints).  If you check the debug messages you'll see a relatively quiet failure.  Or perhaps not.  I think it shows a single success when it should show 4 or more.

In the latest patch I put up, this warning was made much more apparent (a dialog window pops up explaining what happened and redirects the user to the wiki page for troubleshooting steps).  Segfault was also fixed.  Pull and let us know how it goes.  :)

And thanks again for supporting FOSS.  I'm sure the other 1366x768 people out there in the world are sending their thanks to you too.  :P

closed 2018-09-10 by EspoTek

alxpettit, 2018-09-11:
I pulled the latest commit and it works fine! Also, glad to help. :3