espotek-org/Labrador #54: Nice to have; more sophisticated scaling options

drtrigon, 2018-09-11:
As we have the scaling options on the main window now it would be really really nice to polish them a bit.
- up and down arrows should change the decimal place the cursor is (currently it just in- or decrements by 1.0 volt or second which is most of the time way to much)
- may be add sliders or knobs that resemble more the look and feel of a desktop scope (don't know what the GUI toolkit/framework offers)

This is not a bug but a nice to have feature of low priority. Thanks.

EspoTek, 2018-09-11:
I've actually got a class that does point number one already; its the one used in the Signal Generator inputs (which honestly could use a bit of a spit-polish too).  Changing this should be pretty easy.

Honestly not keen on sliders or knobs on the main screen, though.  One of the selling points is that it doesn't have them!  I've used the Hantek software before, which tried to simulate the interface of a "real" scope and honestly it was just really clunky.

EspoTek, 2018-09-17:
Check the latest patch.  The changes I made also improved the signal generator controls and the pop-up dialog.

Let me know what you think.
For now, I'm very happy with everything (although a "time minimum" and "time max" might be better than "time window" and "delay".

I'm closing this issue for now, as I don't think more controls are necessary.  You already have direct range controls on the front page, after all.  :)

closed 2018-09-17 by EspoTek

drtrigon, 2018-09-17:
Thanks for your work! Hmmm - honestly to me it is a bit confusing ... but it might be that I am not used to it yet. It is smoother and makes more sense when you are looking for a signal you don't know the scaling yet. May be a little confusing is the fact the when is written "500mV" I actually have to enter "0.5[ENTER]". Let me work with it for some time... It is clearly better that before.

One thing you should check the behaviour for negative values, crossing "-1.0V" and values below does not work properly. I also had other strange issues. E.g. if I put the cursor into the field "-500mV" (which is default when starting) and hit [ENTER] it changes to "500mV" (positive).

EspoTek, 2018-09-17:
You should be able to type in "500m[ENTER]" to get 500mV.  I just tested it in my machine and that was the case.

Issue with negative numbers noted.  There's an internal variable keeping track of the last known correct input that is not being updated correctly (with sign).  I'll fix this and patch now.

EspoTek, 2018-09-17:
Oh, and one last thing: regarding why you must press enter now:
Say the user wants to enter "500mV".  They type in 5, then 0, 0, m.

If you recalculate the scale on every keypress, it first goes to 5V, then 50V (which goes down to the maximum of 20V because it's invalid), then 200V (which again is reduced to 20V to stay within the valid range), then 20mV.  That is not 500mV.

If you're just trying to get a rough estimate and don't need to be precise, I'd recommend using the mouse scrolling.

drtrigon, 2018-09-18:
> You should be able to type in "500m[ENTER]" to get 500mV. I just tested it in my machine and that was the case.

You are right - guess I entered "500mV[ENTER]" not beeing aware that the "V" stays there and then it jumps to 20V... ;)

Having to press enter is ok and makes sense - I don't like the updates on every keypress (exactly due to issues you describe - some value need more than 1 char to be expressed...)

> If you're just trying to get a rough estimate and don't need to be precise, I'd recommend using the mouse scrolling.

I was not aware of mouse scrolling (now I even got my "knob")! This is great; very useful and fast!!

> I have fixed the issue with negative values. It now uses std::abs() to compare work out the units and scale things, as opposed to the weird hack it was using before.

Thanks for this, however still some non-intuitive things happening here:
- if I press enter in the field the result heavily depends on the cursor position before pressing enter - I assume only the value in front of the cursor is used (e.g. if I put the cursor behind the 5 of -500mV I do get -5V)
- if I start scrolling with the mouse wheel downwards (starting from -500mV) it does 1V steps - I think this is just an issue of a missing init step
- if I start scrolling upwards the same for one step (+1V to 500mV) then it behaves as expected - seems to be sign dependent
- if I try to scroll down from 500mV (passing 0) it stops working at 90mV and the only way to make it work again is entering a value by keyboard

EspoTek, 2018-09-21:
Wow.  And that's why you don't write code on the train after a day of work while sleep deprived!

I've fixed all of the bugs you mentioned, done some more testing to try and find others and it looks like it's all fixed now.  The error regarding the very small (near-zero) number is an issue with floating point precision.  I won't launch into the details, but it's basically a weird, weird form of rounding error.  I've seen an example of it once where a guy managed to get the result 1 + 1 = 3!

drtrigon, 2018-09-22:
What I tested seems to work nicely now even "10us" works - cool, thanks!