espotek-org/Labrador #55: Trigger behaviour

drtrigon, 2018-09-17:
Using following test setup;
- connect signal gen ch1 to scope ch1
- set square 1kHz, 4V, 0V offset
- switch on ch1 and ch2
- switch on trigger (default CH1 Rising, 2.26V)
- set scale to 4V Vmax

The trigger does not behave as expected (by me) in several ways.
- the signals move when pausing or doing a single shot
- the signals move when changing time window (e.g. putting from 10ms default to 11ms makes it appear as if it triggers to the falling edge)
- it is not clear on what edge it is triggered due to that

So may be the main point here is to make clear how the trigger is expected to behave. I do expect it to be set to t=0 and that is clearly not what happens here. Actually I would expect the trigger and t=0 to be in the beginning of the signal or may be somewhere in the center and not at the end.

Interestingly if I "cheat" and pause by switching ch1 to AC coupling (makes the signal not reaching 2.26V and thus the trigger not firing anymore) it behaves as expected from the pause option - it just freezes the signal and does not move it.

EspoTek, 2018-10-29:
Plan on looking at this on the weekend.

https://www.timeanddate.com/holidays/australia/melbourne-cup-day

EspoTek, 2018-11-06:
Looking at this at the moment.  So the way the trigger works is through a software routine that currently works roughly as follows:
 - Each time new data is received, iterate over the samples until you find n sequential samples with a voltage greater than the trigger value (assuming rising trigger; for falling trigger it looks for n samples less than the trigger value).  Call the nth sequential sample "sample X"
 - Calculate the time difference between "sample X" and the front of the buffer.  Call this value "delay x"
 - If the scope is not paused, apply a delay to the signal equal to "delay x" plus TimeWindow/2, to ensure that the sample where the trigger crossing occured is drawn in the centre of the screen.

So that's how it works currently, and I can see a few problems with it:
 - "delay x" should be applied even when the scope is paused.  I've actually already fixed this and it works great.
 - We should not be doing the calculations on "sample X" but instead the first sample to cross the threshold.  Otherwise the frequency of the waveform will affect the position where it is drawn.  This doesn't actually explain the issue with the time window; I suspect that's caused by a different bug that I'm sure I'll uncover._
 - Most people would probably expect the trigger value to be at the left of the screen rather than the centre.

Check the git log over the coming days.  :)

EspoTek, 2018-11-06:
Welp, I lied.  The value "n" in the above algorithm was a function of the window size...

This has been fixed.  
However, there is still a bug where additional delays will be added when the frequency of the waveform cannot be divided equally into the reciprocal of the time window.  Which, to me, sounds like integer rounding issues.  I'll keep digging...

drtrigon, 2018-11-06:
Thanks for your effort! Seems to be a nasty beast... will check as soon as possible!

EspoTek, 2018-11-06:
Haha, I wish I could say it's not a nasty beast, but some parts of the code are much prettier than others.  I don't think the triggering has been touched since about May 2016, and so much has changed around it.

EspoTek, 2018-11-06:
Had another look at it and I think it would be best to just rewrite the triggering code from scratch.  Let me know if you want anything in particular put in.

drtrigon, 2018-11-07:
I far as I can see I need: No Trigger, Single Shot and Continous.

Comming from DSO it would also be nice if the trigger could be moved with respect to the time (default trigger is at t=0 in the center or on the left border). So the trigger should be able to aquire data before and/or after triggering.

Also nice would be the possibility to add voltage offsets to the signals in order to move one channel up, one down for better visibility.

EspoTek, 2018-11-07:
Cool, they should all be pretty easy.  I'll try and get them in soon.

drtrigon, 2018-12-23:
What's the state?

EspoTek, 2018-12-23:
Still in the back of my mind.  Spent most of my Labrador time last month wrestling with website security issues and this month haven't had any time due to Christmas.  About to start making 3kg (!!!) of nougat for the family, and honestly probably won't look at it until the weekend.  Just been dog tired for weeks!  Sorry about that.  I guess that's one of the negatives of (mostly) one-man projects.

drtrigon, 2018-12-23:
Don't mind ... I know all you mentioned except for making 3kg of nougat... ;)
Tell me if I can be of assistance. I also consider I2C testing for you but would like to do both in one shot.

drtrigon, 2018-12-23:
Sorry to hear that! Sadly I do not know anybody currently beeing in Melbourne. That's what I learned years ago when getting familar to pilot plants https://en.m.wikipedia.org/wiki/Pilot_plant "produces small volumes of new technology-based products, mainly for the purpose of learning about the new technology. The knowledge obtained is then used for design of full-scale production" - always do a small volume test run... ;))

EspoTek, 2018-12-23:
Yep, that is usually how we try things out too (and somehow, invariably the pilot batch is tastier), but I needed the nougat today and it had to set overnight.  Good products are rarely made at the last minute, it seems!

robinmh, 2019-01-12:
I downloaded the Dec 2017 version today for the Mac and trigger operates like single shot. Clicking single-shot make no difference to the behaviour.

I think this is an amazing little device (full credit for Mr EspoTek and his tenacity) but I find setting the vertical sensitivity quite difficult and setting the horizontal speed almost impossible.

My requests for consideration would be:

1. Allow the vertical sensitivity to be different between channels (displaying the input and output of an amplifier requires adding an attenuator on the input to make them comparable).

2. Display the horizontal speed in mS per division rather the seconds in scientific notation.

Thanks
Robin

drtrigon, 2019-01-12:
I agree with 1. and would propose for 2. to add an option for horizontal labels to select between what we have or what is used in the range selection part/dialog (SI prefix notation; s, ms, us, ns, etc).

I also think this request is not about trigger behaviour as discussed in this issue and should therefore be added as a new issue about display range/scaling/labeling.

EspoTek, 2019-01-14:
Agreed with 1 but I'd also add adjustable offset in addition to the attenuation.
An addition V/div and mS/div in the top left corner wouldn't hurt either.

EspoTek, 2019-01-28:
Happy Australia Day, Tigon.  :)
https://en.wikipedia.org/wiki/Australia_Day

New triggering algorithm is in place (not perfect yet - doesn't flush state when parameters change and doesn't acknowledge AC coupling), however it does actually stay in the same place when you change the time window!

Once that's been cleaned up, I'll add adjustable offsets.

EspoTek, 2019-02-15:
Have cleaned up the trigger, closing this issue for now.  Will open up a new one with adjustable offsets and attenuation.

closed 2019-02-15 by EspoTek

drtrigon, 2019-02-18:
Will jump in and test as soon as possible. Currently fully occupied with some private stuff (++familiy_size).

EspoTek, 2019-02-19:
Congratulations!  :)

drtrigon, 2019-04-20:
Thanks!

Wanted to test it but the most recent AppImage does not run on my system (kubuntu 14.04):
```
/tmp/.mount_EspoTeCjC0W5/usr/bin/Labrador: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `CXXABI_1.3.8' not found (required by /tmp/.mount_EspoTeCjC0W5/usr/bin/Labrador)
/tmp/.mount_EspoTeCjC0W5/usr/bin/Labrador: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /tmp/.mount_EspoTeCjC0W5/usr/bin/Labrador)
/tmp/.mount_EspoTeCjC0W5/usr/bin/Labrador: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.20' not found (required by /tmp/.mount_EspoTeCjC0W5/usr/bin/Labrador)
/tmp/.mount_EspoTeCjC0W5/usr/bin/Labrador: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `CXXABI_1.3.9' not found (required by /tmp/.mount_EspoTeCjC0W5/usr/bin/Labrador)
```

EspoTek, 2019-04-21:
Hmm, the project uses a newer GCC version now, so perhaps that's the issue.  Can you update to GCC version 7 or later and see if that fixes it?

drtrigon, 2019-04-23:
I would prefer not to. The reason why I used the appimage was that I DID NOT HAD TO INSTALL STUFF to my system...

EspoTek, 2019-04-23:
That's a good idea.  I'll see how best to do it.  Expect a lot of very small .yml file changes tonight.  :)

drtrigon, 2019-04-25:
Cool thanks!

Triggering tested, works now as expected. Other issues appeared...