espotek-org/Labrador #56: Can't connect over USB

AndyWaller, 2018-10-21:
Hi Espotek.

I'm really excited to try my Labrador. However my iMac is unable to detect it. I have tried 3 different micro USB cables, read through the Troubleshooting and installed libusb.Still no joy. 

I also tried to use the Firmware Recovery tool in the help menu with these cables and with and without shorting D1 to GND. The only devices in my home with a micro USB port all use these for charging only (Rpi Zero, a Sperho Mini robot and a rechargeable torch), so I can't know for sure whether these are 
charge only cables or not. I tend to think that all 3 must be charge only cables. 

I'll order a new data cable, but I just wanted to share my experience. Would it be possible to implement a 'cable check' function in software, to help users validate their cables? Although I guess that USB driver issues could also interfere with this.  

Thanks, I'll post what happens with the new cable. 
Andy.

EspoTek, 2018-10-23:
Hi Andy.
Implementing a software cable check is unfortunately not feasible; the USB libraries can see whether or not the device is connected to the USB hub but can't tell the difference between a cable with lines missing and a hardware fault.

I'll leave this issue open for you; just in case you need to do an RMA though.  Let us know how the new cable goes.

~Chris

AndyWaller, 2018-10-29:
Just to update, my proper USB cable arrived I plugged it in and straight away I had oscilloscope traces on the display. All good, thanks. Now to get to work!

EspoTek, 2018-10-29:
Awesome.  Just for what it's worth, since you mentioned you had an iMac, I'm in the process of updating the build scripts to automatically build releases for OSX.  Check the releases page in a couple of days for a new version!

closed 2018-10-29 by EspoTek

EspoTek, 2018-10-29:
Well that was quick.  Check now.  Just uploading it.