espotek-org/Labrador #60: Can't connect on Mac OSX 10.4.1

jherr, 2018-11-23:
I'm getting "Device Not Connected!" on both the CI build and the Update 2 build. I've done "brew update && brew install libusb" followed by rebooting. 

[image]

From the debug console it looks like it does get 'Connected' at some point and then quickly loses connection?

I have no other devices attached, just power. And I've tried two different micro-usb cables.

jherr, 2018-11-23:
I've downloaded Qt Creator to see if I could work on this issue myself. Lots of complaints from the IDE about 10.4 as opposed to 10.3 even with the RC build. When I try to run the cloned code from within Qt Creator I get:

```09:50:59: Starting /Users/jherrr/tmp/Labrador/build-Labrador-Desktop_Qt_5_11_2_clang_64bit-Debug/bin/Labrador.app/Contents/MacOS/Labrador...
QObject::connect: No such slot bufferControl::signalGenIn(bool) in ./ui_mainwindow.h:1573
QObject::connect:  (sender name:   'signalGenGroup_CH2')
QObject::connect:  (receiver name: 'bufferDisplay')
QObject::connect: No such slot timedTickBox::resetTimer(bool) in ./ui_mainwindow.h:1654
QObject::connect:  (sender name:   'lockPsuCheckBox')
QObject::connect:  (receiver name: 'lockPsuCheckBox')
QMetaObject::connectSlotsByName: No matching signal for on_actionRecord_triggered(bool)
QMetaObject::connectSlotsByName: No matching signal for on_actionTake_Snapshot_triggered()
Attempting to open "/Users/jherrr/tmp/Labrador/build-Labrador-Desktop_Qt_5_11_2_clang_64bit-Debug/bin/Labrador.app/Contents/MacOS/waveforms/_list.wfl"
Could not load _list.wfl
```

I've only got:

```
./bin/waveforms/_list.wfl
./build_android/package_source/assets/waveforms/_list.wfl
```

Is there another build step in here that I need to do before I can run the code?

EspoTek, 2018-11-23:
Hi Jack.  On Mac builds only, you'll need to copy the additional directories in Desktop_Interface/bin into the .app bundle.

Are you able to put the board in bootloader mode (See Wiki) and then run the lsusb command from terminal?

If an Atmel DFU bootloader (or similar) appears, you should be able to run Firmware Recovery from the "help" menu in Labrador.app.

jherr, 2018-12-03:
Thanks man, sorry it took me a while to get back to this. Life. Anyway, I think my issue was that I hadn't given permission to the USB driver. ;( Board connects fine now and everything looks great. Thank you! And thanks for this wonderful product. I'll be teaching high schoolers the fundamentals with this and it's so great to have it to teach with!

closed 2018-12-03 by jherr