espotek-org/Labrador #63: Feature Request: FFT / Spectrum Analyzer

drtrigon, 2019-01-09:
Please add the option to view the data on the GUI in frequency domain also. As far as I understand, that basically only needs post-processing the data with FFTW (or similar). This affects the display/view of the data in the GUI only (no changes to firmware needed).

EspoTek, 2019-01-10:
Yeah, this is a good idea.  Simple to add and pretty handy.

stuartc, 2019-02-20:
I would absolutely love this, for diy audio amplifier design and experimentation this would be huge for me!

EspoTek, 2019-03-03:
Het dit, meneer.  :)

I'll need to look into the specifics of how this is done though, specifically how much data should be used per frame and and how often the display should be updated.  To tell you the truth, I've only ever used spectrum analysers for high frequency wireless, where you don't have to worry about the two time periods being on the same order of magnitude.

PWTim, 2020-11-29:
I am jusy finishing my trade course for Télécommunications and i avid capable with a SA VNA OTDR equipment and definitely a low spectrum end Even for good reference thats reliable should be a good extension, 8m sure the tiny SA and NanoVNA are great examples on budget friendly but not quality comprising

sandsmark, 2021-03-05:
fftw is a bit of a hassle to use, I think it would be easier to just pull in this (which also has GPU acceleration): https://github.com/jdupuy/dj_fft/blob/master/dj_fft.h

Vincenzo-Petrolo, 2021-12-02:
I suppose this issue can be closed now, right?

tdunning, 2021-12-30:
Yes. It can be closed, I think since the spectrum display is there.

There are still many issues with the spectrum display, but they can reasonably be dealt with in other issues.  For instance,

- double sample rate apparently suppresses the spectrum analyzer
- can't change the frequency scale
- the trigger status display obscures some of the spectral display
- it would be good to do a sin^2 or sin^4 window on the signal to suppress higher frequencies (only useful if you can change the frequency scale)

Even as this is, it is pretty nice to have.

Vincenzo-Petrolo, 2021-12-30:
Yeah, that's absolutely right. However as you correctly said, another issue should be raised to notify the problem.

tdunning, 2021-12-30:
OK. I will move my suggestions to a new issue.

Vincenzo-Petrolo, 2022-05-21:
@EspoTek this issue can be closed now, since #202 was opened.

closed 2022-05-29 by EspoTek