espotek-org/Labrador #70: Missing calibration data (Windows)

OttoMan28, 2019-03-06:
Every time i start the Labrador windows application, i'm told that the calibration data is not detected.
Its like its not storing the calibration data after a calibration.

[image]

EspoTek, 2019-03-06:
Does this issue still exist when running as admin?  IIRC, it stores the settings in an .INI file in the same directory as the executable (which in theory should be mirrored on Windows to ProgramData but may not be).  

That said, just using a different constructor for QSettings would remove this problem entirely and is a 2-minute fix.  I'll have a look at it when I next have time.

OttoMan28, 2019-03-08:
The 'run as admin' did the trick :-)

Thanks

EspoTek, 2019-03-09:
Hey, don't say "thanks" yet.  That's not a fix, it's a workaround.  :)

Will need to fix the remote builds on Windows and then patch the settings reader before I can accept a thanks.  :P

EspoTek, 2019-03-18:
Well, the hard bit is done here.  Windows remote builds are fixed (barring the old source code versions)!

EspoTek, 2019-04-01:
https://github.com/EspoTek/Labrador/commit/5cc7c9425d70f3c795ba5785a581dcd047a8dd5a

EspoTek, 2019-04-01:
Does that work for you?

EspoTek, 2019-04-23:
Closing this issue now.  Confirmed working on Windows 10 x64 machine.

closed 2019-04-23 by EspoTek