espotek-org/Labrador #77: Errors when running software interface on Raspberry Pi 3

stupid-genius, 2019-04-10:
When attempting to run the software on a RPI 3 (vanilla Stretch), I get the following errors:

    ./res/Labrador: /usr/lib/arm-linux-gnueabihf/libQt5PrintSupport.so.5: no version information available (required by ./res/Labrador)
    ./res/Labrador: /usr/lib/arm-linux-gnueabihf/libQt5Core.so.5: no version information available (required by ./res/Labrador)
    ./res/Labrador: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `CXXABI_1.3.9' not found (required by ./res/Labrador)
    ./res/Labrador: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by ./res/Labrador)
    ./res/Labrador: /usr/lib/arm-linux-gnueabihf/libQt5Gui.so.5: no version information available (required by ./res/Labrador)
    ./res/Labrador: /usr/lib/arm-linux-gnueabihf/libQt5Widgets.so.5: no version information available (required by ./res/Labrador)

I've followed the steps listed in the README.  I have not installed qt-creator, but I also ran the same release on a RPIz and the software seemed to run correctly without qt-creator (the Labrador just wouldn't connect, so I switched to a RPI 3).

EspoTek, 2019-04-10:
Not quite sure why it's demanding specific versions of libs that don't appear on the system.

If you download the source code and follow the build instructions (it's quite simple - 3 commands), does that work?  It should attempt to link against whatever libs are on the system.

stupid-genius, 2019-04-11:
Thanks; I'll try this tonight.

Is qt-creator the only feasible way to build?  Is there no build script I can run from a command line?

EspoTek, 2019-04-12:
You shouldn't need need Qt Creator to build the source code.  You will need Qt5, though.
Just run the commands in the "Building from Source" section on the main page.
https://github.com/EspoTek/Labrador

If you have connection issues, make sure you've run the `setup-udev` script from the Raspberry Pi tarball.  If they still persist, turn it off and on again.

If they _still_ persist, run the firmware recovery script from the `Help` menu in the Labrador software, then disconnect and reconnect the board manually.

Cheers,
~Chris

stupid-genius, 2019-04-13:
`make` fails on RPI 3 (vanilla Jessie) with:

```
g++ -c -pipe -fsigned-char -Wall -W -D_REENTRANT -fPIE -DQCP_VER=1 -DPLATFORM_RASPBERRY_PI -DQT_NO_DEBUG -DQT_PRINTSUPPORT_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_CORE_LIB -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -I. -Iui_elements/qcp1 -Iui_elements -Ibuild_linux -Ibuild_linux/libusb -Ibuild_linux/libdfuprog/include -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-linux-gnueabihf/qt5/QtPrintSupport -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -Imoc -I. -o cursorenabler.o ui_elements/cursorenabler.cpp
In file included from ./unixusbdriver.h:9:0,
                 from build_linux/platformspecific.h:4,
                 from ui_elements/cursorenabler.cpp:2:
./genericusbdriver.h:87:5: warning: identifier ‘nullptr’ is a keyword in C++11 [-Wc++0x-compat]
     QTimer *psuTimer = nullptr;
     ^
In file included from ui_elements/cursorenabler.cpp:1:0:
ui_elements/cursorenabler.h:16:21: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
     bool turnedOn = true;
                     ^
In file included from ./genericusbdriver.h:13:0,
                 from ./unixusbdriver.h:9,
                 from build_linux/platformspecific.h:4,
                 from ui_elements/cursorenabler.cpp:2:
./functiongencontrol.h:18:1: warning: scoped enums only available with -std=c++11 or -std=gnu++11
 enum class ChannelID
 ^
./functiongencontrol.h:28:16: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  double freq = 1000.0;
                ^
./functiongencontrol.h:29:21: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  double amplitude = 0.0;
                     ^
./functiongencontrol.h:30:18: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  double offset = 0.0;
                  ^
./functiongencontrol.h:93:7: error: expected nested-name-specifier before ‘functionGenControl’
 using functionGenControl = functionGen::DualChannelController;
       ^
In file included from ./functiongencontrol.h:11:0,
                 from ./genericusbdriver.h:13,
                 from ./unixusbdriver.h:9,
                 from build_linux/platformspecific.h:4,
                 from ui_elements/cursorenabler.cpp:2:
./xmega.h:20:26: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
 #define INIT_DEVICE_MODE 0
                          ^
./genericusbdriver.h:57:22: note: in expansion of macro ‘INIT_DEVICE_MODE’
     int deviceMode = INIT_DEVICE_MODE;
                      ^
In file included from ./unixusbdriver.h:9:0,
                 from build_linux/platformspecific.h:4,
                 from ui_elements/cursorenabler.cpp:2:
./genericusbdriver.h:58:24: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
     double scopeGain = 0.5;
                        ^
./genericusbdriver.h:59:20: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
     int dutyTemp = 21;
                    ^
./genericusbdriver.h:60:26: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
     bool killOnConnect = false;
                          ^
./genericusbdriver.h:64:22: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
     bool connected = false;
                      ^
./genericusbdriver.h:65:31: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
     bool calibrateOnConnect = false;
                               ^
In file included from ./unixusbdriver.h:9:0,
                 from build_linux/platformspecific.h:4,
                 from ui_elements/cursorenabler.cpp:2:
./genericusbdriver.h:74:25: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
     double psu_offset = 0;
                         ^
./genericusbdriver.h:77:30: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
     unsigned char fGenTriple=0;
[... 65 lines cut ...]
                             ^
./unixusbdriver.h:82:38: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
     int cumulativeFramePhaseErrors = 0;
                                      ^
./unixusbdriver.h:88:25: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
     bool shutdownMode = false;
                         ^
./unixusbdriver.h:89:24: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
     int numCancelled = 0;
                        ^
./unixusbdriver.h:77:68: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11
     libusb_transfer *isoCtx[NUM_ISO_ENDPOINTS][NUM_FUTURE_CTX] = { };
                                                                    ^
./unixusbdriver.h:80:26: error: ‘nullptr’ was not declared in this scope
     worker *isoHandler = nullptr;
                          ^
./unixusbdriver.h:81:29: error: ‘nullptr’ was not declared in this scope
     QThread *workerThread = nullptr;
                             ^
./unixusbdriver.h:99:25: warning: ‘void isoCallback(libusb_transfer*)’ defined but not used [-Wunused-function]
 static void LIBUSB_CALL isoCallback(struct libusb_transfer * transfer){
                         ^
Makefile:651: recipe for target 'cursorenabler.o' failed
make: *** [cursorenabler.o] Error 1
```

EspoTek, 2019-04-13:
You shouldn't be getting warnings about c++11; the project uses c++14.  I'd recommend updating to the latest version of gcc and trying again.

EspoTek, 2019-04-13:
The automatic build scripts use gcc version 7, by the way.

stupid-genius, 2019-04-14:
The version in Jessie's default repo is v4 and is what's installed by default.  Do you have a recommended source for installing a newer version of gcc on RPI 3?

EspoTek, 2019-04-14:
I don't, sorry.  I can look into auto builds for Raspberry Pi, though

stupid-genius, 2019-04-14:
The reason I asked is because you say that it works for RPI, but also that the version of gcc that seems to be required is not what's available in the Jessie repo.  Those two statements seem to be at odds with each other.

Has this been verified to work on a RPI?  What version of gcc was present there?  Where was it obtained (if not from the default repo)?

EspoTek, 2019-04-15:
The binary version provided has been verified to work on both RPi Zero and RPi 3 Model B, using debian stretch and nothing more than `qt5-default` and `qtcreator`.
It was built using gcc-4.9, but the newer code uses a more modern standard that 4.9 is not able to parse.

Honestly, this is the first complaint I've had about it not working, and I can't see any reason why it would throw up the error that it did.

Regarding gcc-7, I had a quick Google and it looks like these instructions should work:
https://solarianprogrammer.com/2017/12/08/raspberry-pi-raspbian-install-gcc-compile-cpp-17-programs/
I've read the shell commands and everything looks like

EspoTek, 2019-04-15:
(Sorry, didn't mean to close!)

EspoTek, 2019-04-25:
Hi again.
I've added a script that automatically downloads Labrador and all necessary compiler tools.
https://github.com/EspoTek/Labrador/releases/tag/raspberry-pi-bootstrap

Tested on a Raspberry Pi 3 running a fresh install of Raspbian stretch.  Please let me know if this fixes the issue for you.

closed 2019-06-23 by EspoTek