espotek-org/Labrador #78: continous android build?

drtrigon, 2019-04-20:
Is there a reason why the most recent android release dates back to end 2017?

EspoTek, 2019-04-21:
Yep, there are no build scripts for it.  Also, a lot of the new features haven't been plumbed into the Android UI.

This is something to work on.

drtrigon, 2019-04-23:
It again the same argument we had some years ago. If the linux binaries need package install I don't like them. So I would like to use the android app, but since there is none... I'm again without working software.

EspoTek, 2019-04-23:
Fair point.  Continuous Android builds will take some time, but I can look into Linux tonight.

EspoTek, 2019-04-23:
Looks like I got close, but no cigar.  Failed on the last step.  I'll try and have a fixed build up ASAP.

EspoTek, 2019-04-24:
Have a build up:
https://github.com/EspoTek/Labrador/releases/tag/continuous-linux-appimage

As linking statically can cause issues on newer systems, I opted to use darealshinji's method here:
https://github.com/darealshinji/AppImageKit-checkrt

I can confirm it runs perfectly on new Linux (kubuntu 18.04 LTS), albeit the interface doesn't look quite as nice as with the native `make install`.  Are you able to confirm that it works on 14.04?

drtrigon, 2019-04-24:
Looks like a nice solution. Tested 23c2fee on 14.04 and it runs. Here is the console output:
```
$ ./EspoTek_Labrador-23c2fee-x86_64.AppImage 
[sudo] password for -----: 
QObject::connect: No such slot bufferControl::signalGenIn(bool)
QObject::connect:  (sender name:   'signalGenGroup_CH2')
QObject::connect:  (receiver name: 'bufferDisplay')
QObject::connect: No such slot timedTickBox::resetTimer(bool)
QObject::connect:  (sender name:   'lockPsuCheckBox')
QObject::connect:  (receiver name: 'lockPsuCheckBox')
QMetaObject::connectSlotsByName: No matching signal for on_actionRecord_triggered(bool)
QMetaObject::connectSlotsByName: No matching signal for on_actionTake_Snapshot_triggered()
Attempting to open "/tmp/.mount_EspoTejbwJuN/usr/bin/waveforms/_list.wfl"
nameBuffer =  Sin

nameBuffer =  Square

nameBuffer =  Triangle

nameBuffer =  Sawtooth

nameBuffer =  DC
newName =  "Sin"
opening "/tmp/.mount_EspoTejbwJuN/usr/bin/waveforms/Sin.tlw"
Length =  128
Divisibility =  4
List loaded!!
Attempting to open "/tmp/.mount_EspoTejbwJuN/usr/bin/waveforms/_list.wfl"
nameBuffer =  Sin

nameBuffer =  Square

nameBuffer =  Triangle

nameBuffer =  Sawtooth

nameBuffer =  DC
newName =  "Sin"
opening "/tmp/.mount_EspoTejbwJuN/usr/bin/waveforms/Sin.tlw"
Length =  128
Divisibility =  4
List loaded!!
Making USB Driver invisible!!
pipeID 0 = 129
Generic Usb Driver setup complete
unixUsbDriver created!
driver =  unixUsbDriver(0x1c7ded0)
axes =  QCustomPlot(0x1b1d490, name="scopeAxes")
CHECKING CONNECTION!
Entering unixUsbDriver::usbInit
Libusb context initialised
DEVICE NOT FOUND
Connected
CHECKING CONNECTION!
Entering unixUsbDriver::usbInit
DEVICE NOT FOUND
Connected
CHECKING CONNECTION!
Entering unixUsbDriver::usbInit
DEVICE NOT FOUND
Connected
CHECKING CONNECTION!
Entering unixUsbDriver::usbInit
DEVICE NOT FOUND
Connected
CHECKING CONNECTION!
Entering unixUsbDriver::usbInit
DEVICE NOT FOUND
Connected
CHECKING CONNECTION!
Entering unixUsbDriver::usbInit
DEVICE NOT FOUND
Connected
dt_userWantsToCalibrate 65536
ReinitUsb entering stage 2

unixUsbDriver destructor ran!
Libusb exited
unixUsbDriver destructor completed!

genericUsbDriver dectructor entering
genericUsbDriver dectructor completed
Reinitialising USB driver!
Making USB Driver invisible!!
pipeID 0 = 129
Generic Usb Driver setup complete
unixUsbDriver created!
ReinitUsbStage2 is returning
ReinitUsb Stage 1 complete
deviceConnectedDisplay::connectedStatusChanged running!
CHECKING CONNECTION!
Entering unixUsbDriver::usbInit
Libusb context initialised
DEVICE NOT FOUND
Connected
deviceConnectedDisplay::connectedStatusChanged running!
CHECKING CONNECTION!
Entering unixUsbDriver::usbInit
DEVICE NOT FOUND
Connected
deviceConnectedDisplay::connectedStatusChanged running!
CHECKING CONNECTION!
Entering unixUsbDriver::usbInit
DEVICE NOT FOUND
Connected
```
(however had no time to confirm https://github.com/EspoTek/Labrador/issues/55 yet...)

EspoTek, 2019-04-25:
Awesome.  I'll have to pass on my thanks to the guy who made the libstdc++ patch.

Ikemeister, 2020-12-11:
Back to Android, when can a build that works for version 11 be expected? Would any error symptoms be useful?

EspoTek, 2020-12-12:
Hi mate, some Android users report no issues, some report no connection at all.
It's something I'd need to discuss with the libusb mailing list, but it should be possible to get out a custom build with better compatibility.

Ikemeister, 2020-12-12:
Thx. Any idea of expected timeframe?

EspoTek, 2020-12-18:
Honestly, can't give an exact timeframe at the moment.  I'll try to have something out around Feb-ish, but can't guarantee.

EspoTek, 2021-03-17:
Now builds on Android thanks to PR from @superfrink.
Still using old libusb, though.

turboencabulator, 2025-05-09:
A continuous Android build is now available.  Please help us test it!

closed 2025-05-09 by turboencabulator