espotek-org/Labrador #80: PSU and Multimeter not working

drtrigon, 2019-04-25:
After solving #55 the PSU does not work anymore. I have to use "Pause" on Scope CH1 twice (Pause on, pause off) in order to make the voltage setting for the PSU work again.

Also the Multimeter shows totaly different values compared to the Scope (the Scope agrees roughly with the PSU settings).

By the way what is the maximum Amplitude for the Signal Generator? I never get more than close to 3.3V. is there a calibration step for the Signal Generator (or am I just to tired and start hallucinating... ;)?

EspoTek, 2019-04-25:
Just tested and it looks like those features are working fine (at least in the normal case).
Here is the test I performed:
```
 - Connect scope CH1 to PSU Vcc
 - Connect scope CH2 to PSU GND
 - Connect board to PC and run the software (EspoTek_Labrador-0c1604c-x86_64.AppImage)
 - Turn on all 4 oscilloscope CH1 stats (max, mean , min, rms)
 - Unlock PSU, set to 8V.  Scope stats show approx 8V.
 - Turn off scope, turn on Multimeter++.  Multimeter++ stats show approx 8V.
```

That said, there are some quirks which are not clearly apparent:
 - Multimeter++ uses CH2 as a reference voltage, whereas the oscilloscope uses GND.  If CH2 is disconnected, it will float to 1.65V, so it's likely that you were measuring the difference between PSU voltage and 1.65V.  (bonus: If you ground CH2, the multimeter can function as a 12-bit scope with amplifier noise calibrated out!  Really handy for taking more accurate measurements.)
 - The Signal Generator uses the same voltage rails as the PSU.  You need to set the PSU voltage to *at least* 1.5V above the peak of the signal generator to prevent clipping.  There is no calibration step for the signal generator, but there's no reason there couldn't be.
 - Also, the PSU has a lock on it.  Are you sure you unlocked it PSU before trying to move the slider?

I'll leave this issue open for now, though.  While I can't say there's a software bug, there's definitely some UI improvements to be made.  The user should be notified of these quirks when they first enable the multimeter or set the signal gen amplitude to a high voltage.

drtrigon, 2019-04-27:
I followed the steps you suggest. After "Unlock PSU, set to 8V" the scope stats stay frozen at around 4.5V. In this case the trick mentioned (pressing "Pause" on scope CH1 twice) does NOT help. I first have to change the range Vmax such that the signal becomes visible, e.g. 10V and THEN press "Pause" twice in order to make it work again.

Here in detail the steps needed to reproduce using EspoTek_Labrador-0c1604c-x86_64.AppImage (on kubuntu 14.04):
1. connect labrador to usb
2. connect CH1 to PSU Vcc
3. start EspoTek_Labrador-0c1604c-x86_64.AppImage
4. answer no when asked about doing a calibration
5. set range Vmax to 7V
already here is obviously something wrong, as the scope trace does not show any noise at all
6. change PSU and see that nothing happens on the scope trace
7. press "Pause" twice and see how it starts working again

I did unlock  the PSU in the first place as the digit changed, actually I assume it's the scope that freezes.

About the quirks you mentioned:
- RTFM to myself ;)) actually it is quite nice to have CH2 as a ref
- why is in DMM mode a scope trace shown? it actually that also has CH2 as ref is that a differential scope mode? same speed as scope? (hope it's not RTFM again ;))

lkoepsel, 2021-02-26:
I have been trying to solve this problem for several days and the "Pause, twice" has fixed it for me as well! So, Thank you, @drtrigon!

Here is what I found out:
1) I have a circuit where I am measuring a potentiometer center connection across a power supply ranging from 0-9.3V DC using the Multimeter function. Due to my circuit needs, I'm measuring in reference to PSU which is 4.8V (measured by an external DMM).
2) This means my measurements will range from -4.8V (0 in ref. to GND) and +4.5V (9.3V of my power supply in ref to GND).
3) When I start the Labrador, immediately uncheck Oscilloscope CH1 and check Multimeter, with the pot positioned at 0V, I have -2.9V on the Max/Min/Mean/RMS MM display. When I rotate the pot to 9.3V, I show +2.94V. Which is incorrect.
4) If I uncheck MM and check both Oscilloscope channels, I am able to see graphically the scope channels are measuring correctly. 
5) If I follow the advice of @drtrigon, press Pause twice on the Oscilloscope channels then switch back to MM, my measurements are correct. I see a range of -4.8V to +4.5V.

lkoepsel, 2021-02-27:
Here's a video of what I am talking about...

https://user-images.githubusercontent.com/7906684/109372524-daea9e00-785e-11eb-877b-ed0cad3a93f5.mp4

EspoTek, 2021-02-27:
I can't see the video - just the audio.

lkoepsel, 2021-02-27:
K, I’ll redo the video. Could you see the problem when you followed the steps outlined?

lkoepsel, 2021-02-27:
Here's a video of what I am talking about...

https://user-images.githubusercontent.com/7906684/109389829-030be880-78c3-11eb-8a7a-a89564d99c43.mp4

lkoepsel, 2021-02-28:
I'm using the Labrador in a classroom and created a presentation for the students to use to overcome the bug. I've also confirmed the bug exists on Linux, Windows and macOS as of the software on Feb 28, 2021. The presentation is attached and might prove more value than the video.

[Labrador MM Bug.pdf](https://github.com/EspoTek/Labrador/files/6057522/Labrador.MM.Bug.pdf)

mi-hol, 2025-02-10:
@drtrigon  could you please let us know if the well documented bug is still present in later versions (i.e. https://github.com/espotek-org/Labrador/releases/tag/continuous-Linux)?

closed 2025-05-30 by mi-hol