espotek-org/Labrador #84: Photo's Suggested on Main Website Product's Page

offtherails2010, 2019-06-07:
Good day to you and well done for making this superb piece of kit thats alas only just caught my eyes !!!!

at 2 inch by 2 inch, its just drop dead gorgeous haha !

Okay on your main website product page, would you be able to show this device, as an example, of it hooked up with the bare bones peripherals needed so one can get a sense of what else is needed to get it off the ground once purchased please please ?

or a good few photo's !!! lol

Once again, totally awesome build, am saving up for one now !!!!

EspoTek, 2019-06-10:
Awesome.  Hopefully you don't need to save for too long!  It was designed to be low-cost and accessible.

I was thinking of putting up a demo video on the site, but just never had the time.  Will hopefully be able to that soon.

eduncan911, 2019-09-28:
I'd also like screenshots of the various user interfaces, specifically Linux and macOS.

EspoTek, 2019-09-30:
Yeah, that's a good idea.  They're mostly the same, but it's best to illustrate that clearly.

closed 2020-09-13 by EspoTek