espotek-org/Labrador #88: Building on Rpi 3 B+ hangs

stupid-genius, 2019-06-23:
While running the bootstrap build script, the build hangs and never completes.  Gets stuck in the same place every time:
`ui_elements/qcp1/qcustomplot.cpp:14952:26`

I successfully built and ran the bootstrap script a few months ago.

stupid-genius, 2019-06-23:
I ran `apt upgrade`and the build finished, but didn't seem to be successful.  Running `labrador` immediately after the build completed was not recognized.

EspoTek, 2019-06-24:
Hmm, that's very odd.  QCustomPlot is a bog standard component that shouldn't have changed at all since I first included it in 2015 or so.  I'm surprised to see an issue there.

I'll have a look at this when I next have some spare time (might be Friday or the weekend) and get back to you with a fix.

One thing to note, are you using the latest version of the script?  It was updated a little while back to fix a missing dependency.

stupid-genius, 2019-06-24:
freshly downloaded

EspoTek, 2019-07-22:
Sorry for the late response.  Have tried reproducing the issue but I'm unable to on my machine.

Could I ask you to try the script again, but change line 29 so that line it reads:
`make CXX=g++-8.1.0 CC=gcc-8.1.0 -j$(nproc)`

Full script below:
[labrador_bootstrap_pi.txt](https://github.com/EspoTek/Labrador/files/3416749/labrador_bootstrap_pi.txt)

smithlevenson, 2019-07-31:
It is hanging for me at nearly the same spot. I added **make CXX=g++-8.1.0 CC=gcc-8.1.0 -j$(nproc)** to line 29. 

I am hanging at: ui_elements/qcp1/qcustomplot.cpp:14952:25

it also says "note:parameter passing for argument of type 'QMap<Double, QCPBarData>::iterator' changed in GCC 7.1" if that makes any difference

The CPU is also pegged 100% every time it crashes. The Pi ACT light flashes every once in a while, but I left it overnight and it was still frozen in the morning.

EspoTek, 2019-07-31:
Sounds like your power supply doesn't appreciate getting hammered during the `make` call.  If you remove the `-j$(nproc)` argument, it'll limit compilation to one core, so you should get past.

smithlevenson, 2019-07-31:
> 
> 
> Sounds like your power supply doesn't appreciate getting hammered during the `make` call. If you remove the `-j$(nproc)` argument, it'll limit compilation to one core, so you should get past.

Thanks! That worked like a charm

EspoTek, 2019-08-01:
Awesome!  I saw that issue too, will create something a little bit more robust on the weekend...

stupid-genius, 2019-08-17:
Sorry it's taken me so long to get back to this.  I edited the script (the one I downloaded before, not the new one you posted in a comment above) to run on a single core and it completed and the app starts correctly.  (I haven't given it a full, functional test yet; this is all I have time for at the moment)

Thanks for the workaround (and the very neat board+software).

closed 2019-08-17 by stupid-genius

EspoTek, 2019-08-18:
Awesome!

I've updated the script just then to include single-core compilation, and also a few other niceties.  There's a one-line terminal command that will automatically download and run the bootstrap script now, and it better handles the case where it terminates part-way through installation.

eduncan911, 2019-09-26:
For the record, I've had dozens of RPi 3s for a project. We bought the cheapest power adapters we could find, and quickly realized a lot of power supplies are fake when putting them under load.

We bought a second batch from another eBay source, and they came in mixed.  Some good, some bad.

All symptoms were exactly described here: couldn't build projects that built fine on other machines, anything taxing the CPU rebooted or locked the device up.  It was madness until we started swapping PSUs.

Finally, we bought the original RPi PSUs, at a huge coat, and have had zero issues since.

RPis are prone to take PSUs and fake SDcards.  9 our of 10 times it's always one of these two things.

EspoTek, 2019-09-30:
I wish I could say I've never done that before.  Pi used to crash every few hours until I changed the microSD card over for a new one.  AU$9 and no problems since.