espotek-org/Labrador #90: Scope XY mode plots in wrong order

emlyn, 2019-06-26:
It looks like in XY mode, the points are plotted in order of increasing X instead of in the order they were captured, leading to the plot getting "filled in" with almost-vertical lines:
[image]

EspoTek, 2019-06-27:
Thanks for the bug report.  Definitely looks like you're right.  Will get onto this ASAP.

Gernreich, 2019-06-27:
Yep saw the same thing when I was making Lissajous figures

EspoTek, 2019-06-30:
[image]

Fixed in 0b007c4.
I was using a "QCPGraph" rather than a "QCPCurve".

As a side note, you can increase the exposure time of the X-Y mode by holding CTRL and scrolling out.  This is a side-effect of the way the plots are made, but it works great.

closed 2019-06-30 by EspoTek

Gernreich, 2019-07-03:
Very cool 😁 thanks!