espotek-org/Labrador #91: Stable release schedule

maxbla, 2019-07-03:
It seems like the last [stable release](https://github.com/espotek/labrador/releases) was 1.1 in 2017. Is there another stable release scheduled?

EspoTek, 2019-07-03:
Don't have any plans for a new stable build, to be honest.  The rolling release has been working pretty well so far.

maxbla, 2019-07-07:
There is an [AUR package](https://aur.archlinux.org/packages/labrador/) that is tied to the stable release, which hasn't been updated since 2017. I suppose a good solution would be to add an AUR package for the CI build (and one that builds from source).

EspoTek, 2019-07-08:
Oh wow.  I had no idea about the AUR package.  Recognise the maintainer, though.

Agreed that an AUR package for the CI build would be nice, but unfortunately I have no idea how to make one.

kdwhite-ufl-edu, 2019-07-08:
Hi Chris,

I wonder if this link might be helpful.  All the Best, Keith

https://www.howtogeek.com/427086/how-to-use-linuxs-ar-command-to-create-static-libraries/

Keith D. White, Ph.D.  Professor Emeritus Department of Psychology  University of Florida
PO Box 112250, Gainesville, FL 32611-2250; 📱 352.219.5863  📠 352.392.7985 • kdwhite@ufl.edu<mailto:kdwhite@ufl.edu>
• https://www.ncbi.nlm.nih.gov/sites/myncbi/keith.white.1/bibliography/53814190/public/?sort=date&direction=ascending<https://urldefense.proofpoint.com/v2/url?u=https-3A__www.ncbi.nlm.nih.gov_sites_myncbi_keith.white.1_bibliography_53814190_public_-3Fsort-3Ddate-26direction-3Dascending&d=DwMGaQ&c=sJ6xIWYx-zLMB3EPkvcnVg&r=FBg6glw4jVkn2Z5pecwiJw&m=g7l0IqoO1xjxWFkeBg8Giej8E_xUsm9rpe8FkHqkpw4&s=dDSYaWMaFHpoaJIToWmiFddGFVAaiq4dkoViKWUulwA&e=>

Associate Coordinator, Brain Function Measurement Core, Brain Rehabilitation Research Center
NF/SG VA Medical Center, 1601 SW Archer Road, Gainesville, FL 32608
Lab: FB-06, D-112, D-116  • 352.548.6000 ext 107675  • kdwhite@ufl.edu<mailto:kdwhite@ufl.edu>
3T MRI Scanner Console • 352.392.5019 (McKnight Brain Institute, LG-114)

Manager, Keith White Consulting LLC, 1205 NW 38th Street, Gainesville, FL 32605-4801
•📠 352.375.6373; 📱 352.219.5863; • kdwhite@gmail.com<mailto:kdwhite@gmail.com>

NOTE: This communication may contain information that is legally protected from unauthorized disclosure. If you are not the intended recipient, please note that any dissemination, distribution or copying of this communication is strictly prohibited. If you have received this message in error, you should notify the sender immediately by telephone or by return email and delete this message from your computer.

EspoTek, 2019-07-09:
Thanks Keith, although that's for a different type of package - `ar`, not `AUR`. 

I guess the real problem, though, is a lack of time.  I currently have a full-time job and second business, in addition to the Labrador project.  I'm getting married in 2 months, too.  

Net result is that I have about 4 hours a week to sit down and dedicate to maintaining Labrador.

I probably could use this 4 hours to research how to make an AUR package, and then get something up and running, and it would honestly be pretty cool, but at the end of the day it doesn't really do much in terms of achieving the project's goal: getting more hardware in the hands of more people and making the on-board tools as accessible as possible.

Additional features, new platforms, bugfixes, documentation and code cleanups have to come first.

Sorry if I'm coming across as a bit dismissive, but when resources are this finite I simply can't do everything.

~Chris

closed 2019-07-09 by EspoTek

kdwhite-ufl-edu, 2019-07-09:
Understood, Chris.  All the Best with your wedding plans!  Keith

Keith D. White, Ph.D.  Professor Emeritus Department of Psychology  University of Florida
PO Box 112250, Gainesville, FL 32611-2250; 📱 352.219.5863  📠 352.392.7985 • kdwhite@ufl.edu<mailto:kdwhite@ufl.edu>
• https://www.ncbi.nlm.nih.gov/sites/myncbi/keith.white.1/bibliography/53814190/public/?sort=date&direction=ascending<https://urldefense.proofpoint.com/v2/url?u=https-3A__www.ncbi.nlm.nih.gov_sites_myncbi_keith.white.1_bibliography_53814190_public_-3Fsort-3Ddate-26direction-3Dascending&d=DwMGaQ&c=sJ6xIWYx-zLMB3EPkvcnVg&r=FBg6glw4jVkn2Z5pecwiJw&m=g7l0IqoO1xjxWFkeBg8Giej8E_xUsm9rpe8FkHqkpw4&s=dDSYaWMaFHpoaJIToWmiFddGFVAaiq4dkoViKWUulwA&e=>

Associate Coordinator, Brain Function Measurement Core, Brain Rehabilitation Research Center
NF/SG VA Medical Center, 1601 SW Archer Road, Gainesville, FL 32608
Lab: FB-06, D-112, D-116  • 352.548.6000 ext 107675  • kdwhite@ufl.edu<mailto:kdwhite@ufl.edu>
3T MRI Scanner Console • 352.392.5019 (McKnight Brain Institute, LG-114)

Manager, Keith White Consulting LLC, 1205 NW 38th Street, Gainesville, FL 32605-4801
•📠 352.375.6373; 📱 352.219.5863; • kdwhite@gmail.com<mailto:kdwhite@gmail.com>

NOTE: This communication may contain information that is legally protected from unauthorized disclosure. If you are not the intended recipient, please note that any dissemination, distribution or copying of this communication is strictly prohibited. If you have received this message in error, you should notify the sender immediately by telephone or by return email and delete this message from your computer.

EspoTek, 2019-07-10:
Thanks Keith.  :)

maxbla, 2019-07-18:
I can package this for the AUR. Generally, AUR build scripts ([PKGBUILD](https://wiki.archlinux.org/index.php/Arch_package_guidelines)s) don't live with the repository, so this wouldn't require any action on your part (i.e. PR, code review). I'd like to give the current maintainer a few weeks to update the current package before submitting a new AUR package.

EspoTek, 2019-07-19:
Oh, thanks so much for the offer.  Much appreciated.  If there's something I can do to make sure it doesn't break further down the track, please let us know.

maxbla, 2019-07-21:
Once there is a AUR package, you could prevent it from breaking by running the AUR build script through your CI. I'm not entirely sure how to do this, right now, but it seems like you could use Travis CI to set up an Arch Linux Docker container, which then runs the build script (using [`makepkg`](https://wiki.archlinux.org/index.php/Makepkg)). [This project](https://github.com/Denton-L/aur-autobuild-template) is an example of the Docker/Arch Linux part. The AUR build script is versioned using Git and hosted by Arch Linux's servers. You could include that git repo as a git sub-module in this repo, then you could make sure you always have the most up to date build script in the CI job.

I can help with all this once there is a more current AUR package.

Maybe you could re-open this issue because there are action items now.