espotek-org/Labrador #93: raspi4/buster, labrador not connecting

edgybudgies, 2019-07-19:
I've been trying to use the labrador on the rpi4 with raspian buster, the software installs, but it can't find the board when I plug it in.  I tried the steps listed in [this issue](https://github.com/EspoTek/Labrador/issues/77), but it still did not connect. I've tried it on an older pi, and a windows machine and it can find the board right away. 

I attached some files, labrador_script_install_log.txt is the output when installing with the rpi bootstrap script, the log.txt is what was outputted, and sudolab is what happens when I launch the app with sudo, unplugging/plugging the labrador in, and running the firmware recovery.

[labrador_script_install_log.txt](https://github.com/EspoTek/Labrador/files/3412824/labrador_script_install_log.txt)

[log.txt](https://github.com/EspoTek/Labrador/files/3412823/log.txt)

[sudolab.txt](https://github.com/EspoTek/Labrador/files/3412825/sudolab.txt)

If you need more info, please let me know!

EspoTek, 2019-07-20:
Out at the moment, can't read through the logs till tonight, but it could be a bandwidth issue.  Does it work when you disconnect all other devices?  If so, try connecting Labrador through a hub.

edgybudgies, 2019-07-20:
When its the only device plugged in, it still can't find it in the GUI. When I launch from the command line, the output loops through these messages over and over:
`CHECKING CONNECTION!`
`Entering unixUsbDriver::usbInit`
`Libusb context initialised`
`Device found!!`
`KERNEL DRIVER INACTIVE`
`libusb_claim_interface FAILED`
`ERROR -5 LIBUSB_ERROR_NOT_FOUND`
`Connected`

`CHECKING CONNECTION!`
`Entering unixUsbDriver::usbInit`
`Device found!!`
`KERNEL DRIVER INACTIVE`
`libusb_claim_interface FAILED`
`ERROR -5 LIBUSB_ERROR_NOT_FOUND`
`Connected`

and when I run `lsusb` it does list an atmel device.

EspoTek, 2019-07-20:
Just had a look over the logs, looks fairly normal apart from the failure to find device.
If you can launch the app at all and initialise `libusb`, you should be able to find the device (when running as sudo, with standard permissions you can have `udev` issues if you didn't install properly).

You mentioned `lsusb` listing at Atmel device.  Which device was it?  The Labrador in "standard mode" (what the software is polling for) has the device ID `03EB:BA94`.  When in bootloader mode, then device will have the ID `03EB:2FE4`.  If the board is stuck in bootloader mode, the Firmware Recovery option (under the `Help` menu) should be able to get the board unstuck.  You may need to disconnect and reconnect the board after running this tool to get everything working again.

By the way, since you've got a Pi 4 and the source code, you may be able to increase the performance to desktop levels!  The software was originally written to run on a Pi Zero.
You'd want to change line 35 of this file here:
https://github.com/EspoTek/Labrador/blob/master/Desktop_Interface/genericusbdriver.h
The lower the number, the higher the refresh rate.  You could probably get away with 33, or even 17.
But either way, probably best to get the board connecting before you fiddle with USB polling rates!

edgybudgies, 2019-07-21:
Seems to be in standard mode: `Bus 001 Device 005: ID 03eb:ba94 Atmel Corp.`

I just reflashed buster again, manually updated usbutils (now 012) and libusb (now 1.0.22), and built it from source again, but that didn't make a difference. I've ran the firmware recovery a few times before and it always says 'command failed'.

EspoTek, 2019-07-22:
Well that throws a spanner in the works...
Walked through the code and cannot for the life of me even begin to imagine what could be going wrong, and don't have the hardware to reproduce this either.

You wouldn't happen to be based in Aus, would you?

edgybudgies, 2019-07-23:
Unfortunately no, different part of the commonwealth. Could be something wonky with USB3 also being on this board now, so don't worry about it. Thanks for all your help and time!

closed 2019-07-23 by edgybudgies

dsmaher, 2020-02-02:
@EspoTek I'm having this issue as well on the RPI 4.  I've got some more info for you:

```$ dmesg
[529943.972869] usb 1-1.4.1: new full-speed USB device number 68 using xhci_hcd
[529944.209203] usb 1-1.4.1: New USB device found, idVendor=03eb, idProduct=ba94, bcdDevice= 2.00
[529944.209222] usb 1-1.4.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[529944.209235] usb 1-1.4.1: Product: Labrador
[529944.209249] usb 1-1.4.1: Manufacturer: EspoTek
[529944.209970] usb 1-1.4.1: Not enough bandwidth for new device state.
[529944.210077] usb 1-1.4.1: can't set config #1, error -28
```
Apparently this is causing the USB interface to be unavailable, so the call to `libusb_claim_interface` fails.  I've poked around and it appears this isn't entirely uncommon on the PI 4 for devices that request `LIBUSB_SPEED_FULL`. 

I haven't found a solution yet, but I'm still looking around for possibilities.  I'd really prefer to use my pi4 if I can for this, since I've essentially made it my "workbench computer".

EspoTek, 2020-02-02:
Thanks for that, Dan.
That really clears explains a lot, and I think I have a solution!

Connect the Labrador through a high-speed (480Mbps) USB 2.0, or USB 3.0 hub.  Do not connect anything else to the hub.
This will prevent the Labrador (or other device requesting Isochronous endpoints at Full-Speed)  from hogging the Pi's bandwidth.

Cheers,
~Chris

dsmaher, 2020-02-03:
Thanks, I have one of these lying around: 
https://www.amazon.com/AmazonBasics-4-Port-USB-2-0-Ultra-Mini/dp/B003M0NURK

And I tried using that, but no luck.  I tried putting that into the USB 2 and USB 3 ports, and neither solved the problem.

I just ordered a USB3 hub to try because I didn't have one available and I'll let you know the result when that comes in.

dsmaher, 2020-02-04:
:disappointed: 
I tried two different USB3 hubs and both of them gave the exact same `dmesg` output that I already submitted :point_up: ... I also tried disconnecting my keyboard & mouse, so _only_ the Labrador was connected to the PI... or only the hub with only the Labrador was connected.  ... and several variations on that theme using different USB ports.

So, still not able to use the Labrador on my PI 4 :sob: ...

EspoTek, 2020-02-07:
Reading more about it, it looks like someone's seriously messed something up internally...

Read P33M's comments here:
https://github.com/raspberrypi/linux/issues/3136

Long story short, there is enough bandwidth to handle the Labrador (through a hub, at least), but either the driver or the hardware has a bug in it where it's unaware of the fact that the hub exists - in violation of the USB spec.

phr0gger, 2020-04-20:
Turns out you can use the Labrador board with the new RPi 4, only not (for now) on any of the USB-A ports.  I powered my RPi 4 via the GPIO pins, added "dtoverlay=dwc2,dr_mode=host" to /boot/config.txt and rebooted.  This enables the hub on the USB-C port and puts it into host mode. Then I connected the Labrador board with a USB-A to USB-C adapter and found it to work without USB errors, even with the web browser open!