espotek-org/Labrador #95: How do I configure Qt Creator on Windows 10?

StarNamer, 2019-07-27:
I cloned the Github source, installed Qt Creator and opened Labrador.pro, but can't seem to get any further. It just keeps saying that the project is not configured and opens the Configure Project tab, but it's not clear what I need to do next. It seems to want either a Kit or a Build, but I don't know where to find these.

I've looked at the Labarador Development Documentation, but that seems to imply it should just open in Qt Creator.

What am I missing?

EspoTek, 2019-07-28:
"the pre-built windows binary is over 2 months old and doesn't have it."
That's a problem!  I'll get it fixed up ASAP.

Regarding the missing kit, Qt Creator is just the IDE and doesn't come with all of the Qt development libs.  They call this a "Kit", and you'll need to install at least one from the Qt Maintenance Tool.  I recommend 5.12.

StarNamer, 2019-07-28:
Thanks. That got me further, although I haven't managed a complete build yet. I'll probably have a further look in a few days.

As you may have deduced, I'm a lot more comfortable with programming in Visual Studio. Although I have been using Linux since 1996, I never got into doing development with it, hence, although I've heard of Qt, I never tried to use it before; any program customisation I needed to do on Linux could simply be done with vi and make!

EspoTek, 2019-08-05:
Hahaha, looks like you're showing your age there.  All the young 'uns now call it `vim`.  `make` is still `make`, though.

Please let me know if you need any more help on this one, or if I can mark it as solved.

JennyEverywhere, 2019-08-05:
Bah. Use a real editor, like Emacs! ;)

(I use nano. I can't stand either vi or emacs.)

EspoTek, 2019-08-05:
If I'm being honest, my most used editor is VS Code.  Have you tried it?  It's amazing.

StarNamer, 2019-08-05:
In the past I've tried Emacs a few times, but never stuck with it. As a mainly Windows programmer, my top 2 these days are Visual Studio and VS Code (which has displaced Notepad++). I usually remote mount a Linux filesystem via Samba and use VS Code on Windows as I don't always have an X server running. I'm usually running Putty to multiple Linux systems, hence the use of vi(m).

P.S. I thought I'd already posted a comment that the new Windows version satisfied my need of a hex output. I haven't had a chance to try building using Qt again yet.

JennyEverywhere, 2019-08-05:
On Linux, I use nano. On Windows, I use Notepad++.

EspoTek, 2019-08-07:
Awesome.  Will close this issue for now, with the following closing note:
Gwen, I implore you to give VS Code a try.  It's wonderful (and FOSS)!

closed 2019-08-07 by EspoTek

JennyEverywhere, 2019-08-07:
I think I have it. I don't do enough coding anymore for it to matter. Most
I do is programming an Arduino now and then.