espotek-org/Labrador #98: Strange display on double sample rate

StarNamer, 2019-09-01:
While looking at a near sine wave, switching on double sample rate gives strange effects, either a flat line for part of the display followed by the expected curve, or even the following, where the curve seems to break up:
[image]
Both scope channels seems to work normally when looking at 2 signals so what's going wrong?

EspoTek, 2019-09-03:
Wow, that's a blast from the past.  This bug haunted me for months back in 2016.

There are two buffers on the microcontroller to hold the ADC's samples.  While one gets filled by the ADC's DMA channel, the other gets emptied by the USB's DMA channel, and they swap back and forth every USB frame.

However, here, it looks like the same buffer is being read from and written to at the same time.

I'll try to get a fix up for this ASAP.

EspoTek, 2019-09-09:
Hi Derek.  I send my apologies.  Simply have not had time to sit down and look at this recently, and probably won't have time this weekend either (getting married!).

I will look at it ASAP after the wedding, though.

~Chris

StarNamer, 2019-09-11:
Hi Chris, 

There's no rush as it's not currently causing me an issue.

Congratulations to you and your wife and I hope you have an enjoyable and memorable day.

Best wishes

Derek

EspoTek, 2019-09-21:
Thanks for the kind wishes, Derek.  Was a wonderful day but very busy!  And, to be frank, I had no idea what I was meant to be doing.  The monotony of normal life makes a lot more sense.

Anyway, back on topic, while investigating this I discovered a second bug - the interface only shows a 33ms (!!) long trace when "double sample rate" is enabled.  This bug has been fixed in the latest commit, however the synchronisation issue you mentioned still remains.

For what it's worth, it seems like disabling and then re-enabling the "double sample rate" mode will prevent the artifacts from appearing 50% of the time (I suspect this has a lot to do with which memory address the DMA channel is writing to when the USB interrupt occurs).  Either way, it's a nasty bug and I'll need to get rid of it!

closed 2022-01-24 by EspoTek