espotek-org/Labrador #99: [Oscilloscope] More inputs

kunaltyagi, 2019-09-03:
Are more signal inputs part of a future plan/release? 4 simultaneous inputs at 8 bits, 750ksps would open a whole new area for me. Or an external clock and trigger to synchronize using an Arduino.

EspoTek, 2019-09-03:
Hi Kunal.

Currently no plans to make an upgraded version in the near future, but when I eventually get around to it (probably years!) I think 4 channels would be a great feature to have.  I suspect it would be more than 750ksps too.

Regarding the external trigger, may I ask what you'd plan to do with that feature?  I'm just not 100% sure why it's needed, given that you've got such a huge sample depth.

kunaltyagi, 2019-09-04:
> you'd plan to do with that feature

I'm trying to focus on a sound sound and nothing else. I was working with 16 bit depth but I think 8 bit is ok too.

Directional mic help in localizing sound and removing noise but they are expensive (and large) and difficult to upgrade. Multiple mic setup is hard with custom electronics. Then I thought, my Labrador already does this. Got a simple setup with dual mic working. Now I want to remove the ambiguities using more mic but getting accuracy is a bit hard with 2 Labradors with independent clocks.

 The trigger/external clock can be used to sync multiple Labradors so that their clocks remain in consensus and it's easy to compare data from different Labradors per timestamp.

> I eventually get around to it (probably years!)

That's disappointing but no pressure. Thanks for the awesome work done till now. I know hardware is hard and I'm totally lost in it.

EspoTek, 2019-09-05:
Hmm, if you're doing audio, it's worth pointing out that the `Multimeter++` mode can operate as a scope with 12-bit samples and noise cancellation (as it shows the difference between CH1 and CH2 in the analog domain).
Of course, the cost is that you only have one channel.

One other thing to note is that the Labrador's clock itself syncs to the USB's SOF packets - so as long as you can make sure they're both being fed SOFs from the same source then the clocks should have the same frequency (at least on average - there will be a small amount of jitter as the SOFs are only sent once per ms).

Easiest way to achieve this would be connecting two USB 2.0 hubs to your computer, and then connecting one Labrador to each.  You'll want to make sure both hubs share the same root hub - ie connect both to USB 2.0 or both to USB 3.0 (not one each); or you could connect the two hubs to a third hub and connect that downstream to any port on your computer.
You'll need to run two instances of the Labrador app as well.

If you exported the samples to CSV and loaded them into Octave then you should sync them up by hand post-fact.  It won't be realtime, but it should work.

kunaltyagi, 2019-09-06:
> you only have one channel

Not only that, but it results in failure to create correlations between the individual signals due to the resultant being a difference of 2 signals. The 12 bit is good, but not good enough to sacrifice one channel. It's a good point to know though.

> connecting two USB 2.0 hubs to your computer

You mean one USB hub with 2 ports? Resulting in the following topology: `Laptop -> USB Hub -> Lab1 and Lab2`

[image]

EspoTek, 2019-09-08:
Nope, I really mean two hubs - a topology like this:
[image]

Most hubs will communicate to non-high-speed devices (such as Labrador) at only 12 Mbit/s.  The result is that  connecting two boards to (attempt to) use between 150% - 200% of the available bandwidth on a single hub (or "root hub" - that's the terminology for the ports on the laptop itself; which usually share bandwidth between one another).
 
However, a high-speed USB 2.0 hub can transmit upstream to the computer at 480Mbit/s - so connecting two boards in the topology pictured uses something like 5% of the total USB bandwidth.

It sounds crazy, but I learned the hard way that that's how 99% of USB hubs (again, including root hubs) work.

kunaltyagi, 2019-09-09:
> It sounds crazy

With USB? Nah. Thanks for the advice. I'll try it out.

closed 2019-09-09 by kunaltyagi