espotek-org/Labrador #100: New addition to the EspoTek team!

EspoTek, 2019-09-17:
Not an issue, just excited to welcome the beautiful Diane-Mari to the Esposito family (and, therefore, by extension, the EspoTek team).

She's the one on the right, in case anyone was wondering.

[image]

JennyEverywhere, 2019-09-17:
Mazel tov, Chris!

gmacario, 2019-09-17:
Congratulations!!!

kdwhite-ufl-edu, 2019-09-17:
> EXTERNAL EMAIL: Exercise caution with links and attachments.

Congratulations!  All the best to both!

Keith D. White, Ph.D.  Professor Emeritus Department of Psychology  University of Florida
PO Box 112250, Gainesville, FL 32611-2250; 📱 352.219.5863  📠 352.392.7985 • kdwhite@ufl.edu<mailto:kdwhite@ufl.edu>
• https://www.ncbi.nlm.nih.gov/sites/myncbi/keith.white.1/bibliography/53814190/public/?sort=date&direction=ascending<https://urldefense.proofpoint.com/v2/url?u=https-3A__www.ncbi.nlm.nih.gov_sites_myncbi_keith.white.1_bibliography_53814190_public_-3Fsort-3Ddate-26direction-3Dascending&d=DwMGaQ&c=sJ6xIWYx-zLMB3EPkvcnVg&r=FBg6glw4jVkn2Z5pecwiJw&m=g7l0IqoO1xjxWFkeBg8Giej8E_xUsm9rpe8FkHqkpw4&s=dDSYaWMaFHpoaJIToWmiFddGFVAaiq4dkoViKWUulwA&e=>

Associate Coordinator, Brain Function Measurement Core, Brain Rehabilitation Research Center
NF/SG VA Medical Center, 1601 SW Archer Road, Gainesville, FL 32608
Lab: FB-06, D-112, D-116  • 352.548.6000 ext 107675  • kdwhite@ufl.edu<mailto:kdwhite@ufl.edu>
3T MRI Scanner Console • 352.392.5019 (McKnight Brain Institute, LG-114)

Manager, Keith White Consulting LLC, 1205 NW 38th Street, Gainesville, FL 32605-4801
•📠 352.375.6373; 📱 352.219.5863; • kdwhite@gmail.com<mailto:kdwhite@gmail.com>

NOTE: This communication may contain information that is legally protected from unauthorized disclosure. If you are not the intended recipient, please note that any dissemination, distribution or copying of this communication is strictly prohibited. If you have received this message in error, you should notify the sender immediately by telephone or by return email and delete this message from your computer.

[70790195_2678428452190043_6052868827245445120_o]<https://urldefense.proofpoint.com/v2/url?u=https-3A__user-2Dimages.githubusercontent.com_22040436_65002295-2D753e7680-2Dd936-2D11e9-2D9524-2D8c02b404d4ae.jpg&d=DwMCaQ&c=sJ6xIWYx-zLMB3EPkvcnVg&r=FBg6glw4jVkn2Z5pecwiJw&m=XheevGFx1zfgn1MTkxHl9I7NuvTyLwP4ER2Rw8xBX6g&s=8ZWYzfQBii-DpDN0RFznoqQVQzGcCV_831lbPHBVo84&e=>

EspoTek, 2019-09-21:
Thanks for all the kind wishes, guys (and gal!).  :)

closed 2019-09-21 by EspoTek

gmacario, 2019-09-21:
Agree to close, this is definitely *not* a bug, but a feature :-)

EspoTek, 2019-09-23:
Hahaha, thanks Gianpaolo.  I showed Diane and it made her day.  :)