espotek-org/Labrador #109: flashing device firmware locks up

perigalacticon, 2020-01-18:
Upon installation of latest interface program on windows 10 pc, and connecting the scope, the program reports "flashing device firmware" but it gets stuck here for at least 15 minutes and appears to be locked up.

EspoTek, 2020-01-18:
I just downloaded the latest version on my machine and plugged in a board with older firmware.  The firmware flash completed in a couple of seconds.

I suspect something else may have changed on your system, or something has happened to the board.
Please restart your PC, and then try the "Firmware Recovery" option from the "Help" menu of Labrador.

perigalacticon, 2020-01-22:
Thanks.  Here's exactly what happens.  If I plug in the board before starting the program a notification window pops up asking for calibration.  It doesn't matter what I choose, the program then fails to start.  When launch it again (from my desktop) it asks to calibrate, again my choice doesn't affect the result but then it starts and shows device not connected.  If I then re-plugin the board it goes to flashing device firmware, and nothing I do changes that message and the scope does not run but the controls seem to react very slowly as if it is bogged down doing something.  

If I launch the program before connecting it still asks to calibrate, again my choice does not affect it and the only difference is the program launches the first time.  

I tried the firmware recovery help option, it says it appears to be configured so the problem could be the USB set up.  So I plugged the USB into various USB ports, including a USB hub I have that has the blue USB ports mentioned.  Nothing has changed the behavior of the launch process I described.

EspoTek, 2020-01-27:
Odd.  Did it used to work in an older version of the software?

Under `C:\Program Files\EspoTek\EspoTek Labrador\firmware` you should see a batch file called "flash.bat".  If you set the board to bootloader mode (see https://github.com/EspoTek/Labrador/wiki) and then run that batch file, does it fix things?

phr0gger, 2020-04-20:
I have seen Labrador showing a (bogus?!) Flashing Firmware message on the Raspberry Pi 3B+. 
It happens when restarting the Labrador software (but not unplugging the board) after there were -9 errors (unixUsbDriver::usbSendControl FAILED with error LIBUSB_ERROR_PIPE). The board supposedly reports FIRMWARE VARIANT = 0x00 and the software attempts to flash the board. However, the USB driver is still not working and nothing happens. One needs to power cycle the board to re-initialize the USB driver.

99Phish, 2021-09-12:
Terribly sorry for resurrecting an old thread.

Im afraid I am in the same boat as op.  I have the software running on my windows work pc, Albeit an older version. I also have it running on macOS after the recent continuous build update (the one after 9/2/21).

I am now just installing the software fresh on my personal pc only to encounter "flashing device firmware" without resolution..

This board has saved my hide in the field more than once so I am not complaining.  I just thought I should document the problem. I also tried plugging in the board before and after I start the software. All to the same effect. 

On my work pc and my work macbook, sometimes I need to unplug and replug the usb cable to get the software to recognize the controller on boot up.  After a single power cycle it always gets recognized and hums along as well as expected.

I seem to be running into a driver problem based off of device manager. The first image "flashing" is what I encountered after plugging in the device.

The second image "reinstalled" is what I encountered after reinstalling the current windows build. 

 [image]

[image]

--later on--

I happen to have my macbook pro with me and  connecting the labrador to that computer also results in failure. This time I am being prompted to connect the device.  

The problem is likely on my end all things considered.  I typically keep the board in my backpack in the original packaging. Pink foam and the esd bag.

EspoTek, 2021-09-12:
Hi mate, that's very bizarre.  I suspect it's a hardware fault, but it could also possibly be a bad flash.

The first thing I'd do is to see if the board enumerates correctly when in bootloader mode.
To do this, connect Digital Out 1 to GND then reconnect your board:
https://github.com/EspoTek/Labrador/wiki/Pinout

You should see an "Atmel ATXMega32A4U" device in Device Manager if the bootloader is good.

If that works, you can go to C:/Program Files/EspoTek/EspoTek Labrador/Firmware and run the files "wipe.bat" then "flash.bat".  They should take a couple of seconds each to run (if the window immediately closes, that's not good) and the board should be back to normal.

If that doesn't work, I'd recommend trying a different cable.  If it's not the cable, contact me at admin@espotek.com and we can look into getting a replacement sent to you.

Cheers,
~Chris

closed 2021-09-12 by EspoTek