espotek-org/Labrador #110: Documentation update request

MatthewSteeves, 2020-01-21:
New to using the EspoTek Labrador and have compiled this list of documentation updates that would've made my ramp-up easier and quicker.

I didn't want to flood you with issues, so I've just lumped the list together and placed in (my own) order of importance/preference. You may disagree. :) 

My intent is not to burden you with a bunch of work - my belief is that most of these doc updates are a few sentences or less.

1. Update Signal Generator documentation to say that max sig gen voltage depends on PSU voltage (or update code to interlock/auto change). I had assumed they were not inter-linked based on the first bullet point of [the tutorial](https://github.com/EspoTek/Labrador/wiki/Tutorial:-Connecting-a-Simple-Circuit-to-the-EspoTek-Labrador).
2. Clarify PSU On/Off jumper position on [pin out diagram](https://github.com/EspoTek/Labrador/wiki/Pinouts) (The order of text in the diagram is actually opposite of the jumper setting)  (even better if on phys board for future batches)
3. Document reason for existence of TWO oscilloscope Duplicate pins per channel
4. Provide PDF of schematic (and make accessible from Wiki) so that don’t have to install Kicad or go searching the web (I found a copy [here](https://raw.githubusercontent.com/tardate/LittleArduinoProjects/master/Equipment/Labrador/assets/Labrador_schematic.jpg))
5. Document ESC key (resets Labrador?)
6. Copy board specs from website to GitHub wiki (I often want to reference them, and usually have the wiki open if using the board.)
7. Clarify Oscilloscope grounding procedure (do both have to be done simualtaneously?) and that can use PSU ground and other ground pin instead
8. Doc ammeter burden voltage
9. Doc intent of PSU expansion header
10. Doc intent of FUSE bypass (ie when would you use it / risks/rewards of doing so)

EspoTek, 2020-01-27:
These all seem like pretty sane recommendations.  The documentation is years out of date and needs some more work.  If anyone else has some thoughts, please feel free to share them in this thread too.

phr0gger, 2020-04-17:
Why is editing not enabled for the WiKi?
Maybe the community will fill in? For instance I would add the schematic after going through the trouble of installing Kicad just to get at it.

EspoTek, 2020-04-20:
The Wiki is now publicly editable.

skorokithakis, 2020-12-08:
These questions are questions I have too, plus I'm wondering what the "digital out" pins do, and I haven't been able to find information on a cursory search.

mi-hol, 2025-02-10:
@MatthewSteeves I'm trying to help administrate the project and improve the project.
Help from the community is greatly appreciated.

Would you mind the review the list of doc updates you provided and open new sub-issues for the ones remaining

At best maybe you could update topics you are familiar with directly in the wiki pages?