espotek-org/Labrador #112: Images not working in wiki?

msabramo, 2020-01-30:
It seems like images, such as the pinout and photo of use with a breadboard are not rendering for some reason.

msabramo, 2020-01-30:
Hmmm the screenshot I just took isn’t showing up either. Seems like maybe GitHub is having a problem with images?

EspoTek, 2020-01-30:
It's loading just fine for me.  Maybe try a different browser, or private mode.

Either way, I'll close this issue because there's nothing I can do about it - I don't have any control over GitHub's servers!

closed 2020-01-30 by EspoTek

msabramo, 2020-01-30:
Thanks, Chris!

I solved the problem.

For any other folks who happen to run into this problem, I'll explain the situation...

I am in the U.S. (San Jose, California in particular) and my Internet provider is Comcast/Xfinity. I noticed that I was having this image loading problem on all of the computers on my home wifi network and all phones as well, so since Chris said it worked fine for him, I suspected it was related to my home network. Based on that hypothesis, I disconnected my phone from wifi and let it use 5G. Then the images worked on that device. So that bolstered the theory that it was caused by the home network. I then opened the image in a new tab on my computer and got an error page and noticed that the favorite icon for the tab was an Xfinity logo. I then tried hitting stuff from the command line with `curl` and saw this:

```
$ curl -sS user-images.githubusercontent.com | grep -i 'protected browsing'
			<p class="opacity__70">Protected Browsing is enabled and <span class="br">has blocked this content because it's suspicious.</span></p>
			<p class="opacity__70">To disable this setting, go to the Xfinity xFi mobile app or <a href="https://xfinity.com/myxfi" target="_blank">xfinity.com/myxfi</a> <span class="br">and navigate to Network, then select Protected Browsing.</span></p>
```

so based on that, I opened the xFi app on my phone and browsed to the network. Then oddly while navigating around in there, the app popped up a message that said "Advanced Security enabled". I then refreshed the page on my computer and it worked!

TL;DR: If images aren't loading for you and you happen to have Xfinity internet service, open the xFi app on your phone and see if that fixes your problem.

EspoTek, 2020-01-30:
My God.  Welcome to the future...