espotek-org/Labrador #113: Auto-ranging broken on Multimeter++ mode.

EspoTek, 2020-02-04:
Have had a couple of users complaining about their scope being unable to go above 3V.  Checked in the software today and it looks like the PGA is being configured incorrectly.  I should be checking for a stream of `FFF` and bumping up the range.  Would probably want to bump it down, too, if every sample is below some threshold (`9FF`?)

mff47025, 2020-11-20:
Has this issue been resolved?  I just got a new board today and the scope clips at 3V.

mff47025, 2020-11-20:
> Has this issue been resolved? I just got a new board today and the scope clips at 3V.

Ok. I see the issue now.  I can get the signal generator above 3V, if I turn the Power supply up.  I was able to get a 5V sine wave out by turning the power supply up to about 9 or 10V.

skorokithakis, 2020-12-13:
I think my issue is related: To generate a 5V signal in the generator, I need to bump the PSU up to 7.25V. Is this the same issue? I can live with it fine, I just want to confirm that I'm not doing something wrong.

skorokithakis, 2020-12-13:
Actually, it looks like this is a bit more annoying than I initially thought. Because I'm powering a circuit with 5V from the power generator and want to use a signal generator to analyze what happens in another part when the voltage goes from 0 to 5V, I'd have to give the circuit 7.5V for the signal generator to go up to 5, which would potentially harm my circuit.

@EspoTek would it be possible to fix this? It would be quite appreciated.

EspoTek, 2020-12-14:
Hi Stavros, it's impossible to solve this without adding hardware - there is a single power rail which drives both the signal gen amplifier and the power supply output.

It is possible to solve, though!

The neat solution would be to add a 5V regulator after the Labrador's Power Supply, the minimal, more fun solution would be to connect your circuit directly to the +5V USB line (either by scraping off part of the solder mask on the board and soldering on a bodge wire, or splicing the red line in your USB cable).

skorokithakis, 2020-12-14:
Ahh sorry, I thought it was a software issue. I think I'll go with the USB pad alternative, thanks Chris!

skorokithakis, 2020-12-14:
@EspoTek can I ask what causes the problem, though? Theoretically, with the power supply at 5V, the sign generator should be able to go up to 5V too, not just 3V, correct?

EspoTek, 2020-12-14:
@skorokithakis, The LM324 (op-amp used to drive the signal generator) has a maximum output voltage around 1.5V (two diode drops) lower than its supply voltage.  When you get close to that Vcc-1.5V range, it goes nonlinear.
This is typical for many op-amps.

skorokithakis, 2020-12-14:
Ahh okay, thank you. The 5V workaround seems good enough, thanks!

skorokithakis, 2020-12-14:
Sorry to post on the wrong issue again, but, in case someone is interested, you do expose a 5V pin straight from the USB in the "fuse bypass" pins:

[image]

I'll just use that, maybe it would be helpful if it were added to the pinout as well (it's a very useful pin in general).

EspoTek, 2025-02-12:
I think it's probably better to close the issue off - this is obsolete now!

closed 2025-02-12 by EspoTek