espotek-org/Labrador #121: taking samples for both channels simultaniously via librador

jhegedus42, 2020-06-26:
Hi,

I am able to get samples from a single chanell using a single octave call via librador.

Now, the question is ... how can I get samples for two channels ?

I am trying to understand this : https://github.com/EspoTek/Labrador/blob/e7c5e4e68768f87d1df8cc61be07e32e3b8d232e/Librador_API/___librador/librador_shared_library/o1buffer.cpp

and this : 

but its not easy ... 

https://github.com/EspoTek/Labrador/blob/e7c5e4e68768f87d1df8cc61be07e32e3b8d232e/Librador_API/___librador/librador_shared_library/usbcallhandler.cpp

maybe if I could modify this : https://github.com/EspoTek/Labrador/blob/e7c5e4e68768f87d1df8cc61be07e32e3b8d232e/Librador_API/___librador/librador_shared_library/usbcallhandler.cpp#L313

to do something like this : https://github.com/EspoTek/Labrador/blob/e7c5e4e68768f87d1df8cc61be07e32e3b8d232e/Librador_API/___librador/librador_shared_library/usbcallhandler.cpp#L76

but what is isocallback ? I am really sorry for all these dumb questions but C/C++ is something I never was able to 
lear :( ... I try to read the code... but it's difficult.

can someone please point me into the right direction ? Please ?
 
I am really lost here ... :(

Thanks,

Jozsef

phr0gger, 2020-06-27:
I have some python code to configure the Labrador board and to read back data.
Would that help you?

EspoTek, 2020-06-29:
When the device is set to mode 4, the latter 375 bytes of the each packet contain the data for CH2.
https://docs.google.com/document/d/1ZDO1RTarQTNB6Pdfi_T4YukL51oYan_kk_teb1cu6-o/edit#heading=h.3v1s22knpngz

jhegedus42, 2020-07-15:
> I have some python code to configure the Labrador board and to read back data.
> Would that help you?

Thank you that would be great, is it on github ? I check your code now.  Maybe it is already there.

phr0gger, 2020-07-16:
I don't think my code is GitHub-worthy.
Below is what I did on my Raspi to read and plot data from the Labrador board with Python 3.7. It is based on the sample code by Vincent Pelletier, who wrote the python-libusb1 package which you need for the asynchronous support.
PyUSB does not support asynchronous isochronous transfers.

`import usb1  # python-libusb1 package from 
from struct import unpack
import sys
import time
import numpy as np
import matplotlib.pyplot as plt

MODE = 0  # Labrador mode ( 7: 12 bit 375 ksps, 6: 8 bit 750 ksps, 0: 8 bit 375 ksps)
SHOW = 20*750

VENDOR_ID = 0x03eb
PRODUCT_ID = 0xba94
INTERFACE = 0
TIMEOUT = 100

GAIN = 0x0808  # gain 4: ±6V
#GAIN = 0x0c0c  # gain 8: ±3V

ISO_EP = 0x81
NUM_ISO_PACKETS = 66
ISO_PACKET_SIZE = 750
TRANSFER_COUNT = 4

def processReceivedData(trf):
    global stream
    global t0
    if MODE == 7:
        dt = np.dtype(np.int16)
        dt = dt.newbyteorder('<')  # big endian
    else:
        dt = np.dtype(np.int8)
    t1 = time.monotonic()
    if trf.getStatus() != usb1.TRANSFER_COMPLETED:  # TRANSFER_COMPLETED = 0 
        # Transfer did not complete successfully, there is no data to read.
        # This example does not resubmit transfers on errors. You may want
        # to resubmit in some cases (timeout, ...).
        print(' Error: Transfer not completed successfully!')
        return
    stream.extend( np.frombuffer(trf.getBuffer(), dtype=dt)) 
    #stream.extend( trf.getBuffer())  # raw hex stream
    # trf.getActualLength() useless for isochronous (always 0)
    print(f' time: {1000*(t1-t0):3.2f}')  
    # Resubmit transfer once data is processed for endless process
    #transfer.submit()
    t0 = t1

stream = []
with usb1.USBContext() as context:
    context = usb1.USBContext()
    handle = context.openByVendorIDAndProductID(
        VENDOR_ID,
        PRODUCT_ID,
        skip_on_error = True,
    )
    if handle is None:  # Device not present, or user is not allowed to access device.
        print('Error: Cannot find Labrador Board!')
        sys.exit()
    else:
        print('Found', handle.getManufacturer(), handle.getProduct())
        print(' FW Version :', unpack('H',handle.controlRead(0xc0,0xa8,0,0,2,TIMEOUT))[0])
        print(' FW Variant :', handle.controlRead( 0xc0, 0xa9, 0, 0, 1, TIMEOUT)[0])
            
    with handle.claimInterface( INTERFACE):  # Required to receive/send data.
        handle.controlWrite( 0x40, 0xa5, MODE, GAIN, b'', TIMEOUT) #
        handle.controlWrite( 0x40, 0xa6, 0, 0, b'', TIMEOUT)  # turn off digital pins
        handle.controlWrite( 0x40, 0xa4, 0, 0, b'', TIMEOUT)  # disable DAC amp
        handle.controlWrite( 0x40, 0xa3, round(6/18.15*128)-1, 0, b'', TIMEOUT)  # set 6V

        # Build a list of transfer objects and submit them to prime the pump.
        transfer_list = []
        for i in range( TRANSFER_COUNT):
            print( ' Starting transfer',i)
            transfer = handle.getTransfer( NUM_ISO_PACKETS)  # libusb_alloc_transfer
            transfer.setIsochronous(                         # libusb_fill_iso_transfer
                ISO_EP,
                NUM_ISO_PACKETS*ISO_PACKET_SIZE,             # 
                callback = processReceivedData,
            )                                                # default is no timeout
            transfer.submit()                                # libusb_submit_transfer
            transfer_list.append(transfer)
        print()
        t0 = time.monotonic()
        # Loop as long as there is at least one submitted transfer.
        while any(x.isSubmitted() for x in transfer_list):
            try:
                context.handleEvents()                      # libusb_handle_events
            except usb1.USBErrorInterrupted:
                pass
if MODE == 7:
    data = -np.right_shift( stream, 4)[:SHOW]
elif MODE == 6:
    #data = list(map(lambda x: -x, stream[:SHOW]))
    data = stream[:SHOW]
elif MODE == 0:
    data = np.reshape(stream[:2*SHOW],(4*SHOW//750,375))[::2].ravel()
fig, ax = plt.subplots(figsize = (12, 7))
ax.plot( data, '.', markersize = 2)
ax.set_xticks(range(0,len(data)+1,375))
ax.grid()
ax.set_title( f'Mode: {MODE}, Gain: 0x{GAIN:04X}')
fig.tight_layout()
plt.show()  
`

jhegedus42, 2020-07-16:
Thank you !

It seems to work !

[image]

jhegedus42, 2020-07-18:
> I don't think my code is GitHub-worthy.
> Attached is what I did on my Raspi to read and plot data from the Labrador board with Python 3.7. It is based on the sample code by Vincent Pelletier, who wrote the python-libusb1 package which you need for the asynchronous support. Note, that PyUSB does not support asynchronous isochronous transfers, read and write are blocking.

Would it be ok to use blocking calls?

C itself is blocking - if using Librador.

Why is it good to use async lib?

I mean, why is it bad if one python process is blocking?

I plan to use TCP/IP to transfer packets read from Labrador to Simulink/GNURadio, that can be blocking, python runs in separate process.

I did this already with UDP packets using Librador library from C. It worked for one channel. Using python seems to be easier though.

But do I need async calls for that?

It just worked fine with Librador + blocking calls to libusb + C + UDP + Simulink + FIFO (buffer).

Is async USB transfer with callbacks any better than blocking calls ?

It seems more complicated.  

Is the Labrador QT GUI using async USB calls or just using blocking USB calls + buffer + 2 threads (one for GUI and one for USB transfers).

Cheers,

Jozsef

EspoTek, 2020-07-20:
I think I can answer that question, Joszef.

The GUI code uses async isochronous USB (but synchronous everything else), and the main advantage of this is no data loss when streaming.
The board itself only stores the previous 1ms of data in SRAM, so in order to extract every sample it must be polled every single USB frame - no exceptions.
This is impossible to guarantee with synchronous communications, as there is some downtime between packets being received and packets requested.
The flow looks something like this:
 1) Request _n_ frames of data
 2) Data received 
 3) Process data
 4) Goto 1
If steps 2 and 3 take longer than 1ms, you lose continuity.

With async, you can immediately queue a second (future) request after the first request is submitted.  Then, when the first request is complete, you can schedule a third before the second completes (and so on...).  This way you don't lose continuity.

Try looking at an odd frequency (I like 912Hz) and you'll see what I'm talking about.  You'll get a discontinuity every `NUM_ISO_PACKETS` samples.

(Apologies if that made no sense.  I just became a father on Saturday and haven't had a nights' sleep since)

jhegedus42, 2020-07-20:
Dear Chris,

First of all. I wish you and your family health and happiness !

Thank you for the help. It is very good to know this. It's not trivial to make such a 
little device work. I start to understand that.

I rather ask now @phr0gger ... he may not be so busy now.

Is there a way to make this record data continously ?
The python code that you attached, @phr0gger  ?

What does it mean 
"  # Build a list of transfer objects and submit them to prime the pump."

"prime the pump" ?

Do you mean bootstrap ?

I was thinking I should uncomment this line : 

```
    # Resubmit transfer once data is processed for endless process
    #transfer.submit()
```
but it just died...

or did not really work...

what should I do if I want to continously capture data and, say, send them over
TCP/IP or just save to file ?

Cheers,

Jozsef

closed 2020-09-13 by EspoTek