espotek-org/Labrador #122: Device driver not installed

iryek219, 2020-07-03:
I cannot connect the board on Windows 10 64bit, Windows 7 32bit, and Ubuntu 18.04 64bit.
Both Windows systems are based on Intel cpu and Ubuntu AMD.

On connecting the board through USB, I see that driver of 'USBGobindar dummy device' is not installed.
The hardware id of the device is like below:

    VID_03EB&PID_A000&REV_0200
   
Please help me setup the board.

EspoTek, 2020-07-04:
Can you please provide the output of lsusb on the Ubuntu system, before and after connecting the board?

iryek219, 2020-07-04:
Before connecting the device

hwan@hp-g6:~$ lsusb
Bus 002 Device 003: ID 0c45:6458 Microdia 
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

After

hwan@hp-g6:~$ lsusb
Bus 002 Device 003: ID 0c45:6458 Microdia 
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 024: ID 03eb:a000 Atmel Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

EspoTek, 2020-07-05:
This is a serious error.  It looks like the latest batch of boards were misconfigured at the factory.
I'm patching the software now to detect and rectify this.

EspoTek, 2020-07-05:
Please download the latest version of the software (released today).  It has a new function in it that will detect the configured boards and automatically repair them.

Thank you for sharing the bug report.
Please let me know if this does not fix the issue.

closed 2020-07-05 by EspoTek

iryek219, 2020-07-05:
Let me confirm that the latest release works. Now, the device connects without any problem. And, I've done the simple tutorial of voltage divider. 
Thanks for your quick actions!!!

EspoTek, 2020-07-06:
Not a worry.  The fix wasn't too difficult, just needed a customer to explain the problem.  So thank you for your report.  :)