espotek-org/Labrador #123: Driver not installed

sanjivvenkatram1972, 2020-07-04:
Dear team,

Greetings, hope you are well and staying safe.

I installed Labrador on my Windows 10 PC (64 bit). I see the following issue and kindly help.

1. Upon connecting the board through a data cable (recognizes my phone), the software says device not connected. 
2. I looked at device manager (was in "other devices" as "USD Gobindar Dummy Device". 
3. Opened this and tried updating driver by "picking from the list" and unfortunately don't find "libusbK Usb Devices".
4. Re-installed software and made sure both driver boxes were ticked.
5. Finally, clicked on firmware recovery on Labrador's help menu and following it through got a message saying "labusb0.dll was not found".
6. Checked the drive and so see this file.
7. Also ran "flash.bat" and unfortunately don't see any change.

Kindly help as I need to for customer meeting on Monday :-).

Kind regards,

Sanjiv

EspoTek, 2020-07-05:
Duplicate of #122

EspoTek, 2020-07-05:
(Check your email, though, responded there with a fix!)

closed 2020-07-05 by EspoTek